vlachoudis/bCNC #1336: Python3 quirks with clipboard at startup

onekk, 2020-01-08:
An error is generated in Python3 when pasting using the editor paste button at startup (at startup there is no data in the clipboard)

```
    obj = unpickler.load()
TypeError: a bytes-like object is required, not 'str'

```

relative to the paste method on CNCList

```
        try:
            # try to unpickle it
            unpickler = pickle.Unpickler(StringIO(clipboard))

            try:
                while True:
                    obj = unpickler.load()
```

I suspect that it is caused by new way of treating pickle objects in python3 and that StringIO is wrong  as python3 unpickler expects bytes as inputs

some more error appear when trying to copy as even in CNCList copy 

```
    def copy(self, event=None):
        """Copy selected items to clipboard"""
        sio = StringIO()
        pickler = pickle.Pickler(sio)
        # sio.write(_PLOT_CLIP)
        for block, line in self.getCleanSelection():
            if line is None:
                pickler.dump(OCV.blocks[block].dump())
```

at 

```
   pickler.dump(OCV.blocks[block].dump())
TypeError: string argument expected, got 'bytes'

```

pickle is changed between python2 and 3 so maybe a complete rewrite is need for the clipboard to work as expected.

Regards
Carlo D.

Harvie, 2020-01-08:
You probably have old bCNC version. AFAIK this was fixed in #1328

onekk, 2020-01-08:
Trying the last build from github of bCNC here some error

```
new-config Utils <ConfigParser.ConfigParser instance at 0x7f3ae7d95200>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 52428890
Traceback (most recent call last):
  File "/home/carlo/devel/bCNC-master/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/home/carlo/devel/bCNC-master/bCNC/__main__.py", line 705, in paste
    self.editor.paste()
  File "/home/carlo/devel/bCNC-master/bCNC/CNCList.py", line 210, in paste
    objs = json.loads(clipboard)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

```

python3 seems to work, at least with 3.5.3 from my devuan ascii (debian stretch) installation.

DodoLaSaumure, 2020-01-09:
@onekk Does the problem persit after this commit 3cee948 ?
Thanks forward for sharing the info
Pierre

onekk, 2020-01-09:
with python 3.5.3 on linux, Devuan ascii same as Debian 9 but without systemd. clicking on paste at the startup gave this error:

```
Traceback (most recent call last):
  File "/home/carlo/devel/bCNC-master/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/home/carlo/devel/bCNC-master/bCNC/bCNC.py", line 705, in paste
    self.editor.paste()
  File "/home/carlo/devel/bCNC-master/bCNC/CNCList.py", line 210, in paste
    objs = json.loads(clipboard.encode("utf-8"))
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'

```
Python2 (2.7.13) on same OS raise this error instead:

```
Traceback (most recent call last):
  File "/home/carlo/devel/bCNC-master/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/home/carlo/devel/bCNC-master/bCNC/bCNC.py", line 705, in paste
    self.editor.paste()
  File "/home/carlo/devel/bCNC-master/bCNC/CNCList.py", line 210, in paste
    objs = json.loads(clipboard.encode("utf-8"))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

```
Maybe the problem is not emptying the clipboard at startup, or making some checks to verify the object tyopes on the clipboard, eventually sorting out the different types of "valid" objects, I could guess they could be ascii strings (Gcode if I didn't go wrong use "only" strict ascii strings) and eventually Blocks() objects "serialized" as JSON, at least for the paste method (In theory copying from bCNC will copy only "valid" objects), or implement a internal clipboard method,   instead of using the System clipboard, I couldn't think of other methods, maybe the checking is more easy to implement, simply rejects (maybe with an error windows) the illegal objects.

As a side note, I'm developing my very taylored version of bCNC, with a similar codebase for the core part, (taylored to use my favourite CAM generated files, so for simplicity is bCNC with some custom functions and all the CAM operations stripped out, optimizations is not needed, and the Gcode parser for creating Block() are highly customized for the CAM I use (CamBam) to detect the Machine Operations (MOP) and when finished the shapes in each MOP).
So I could not be much in help as I didn't use bCNC daily (for now is literally as the machine is in a room wothout heating and here (Italy) is winter).

Testing is done with a Arduino Nano with GRBL on it set up with no hard limit switch so it could be used as a cheap simulatore (a clone of the Nano could cost less than 5€)

Regards

Carlo D.

DodoLaSaumure, 2020-01-09:
Thank you @onekk for the clarifications.
I understand that you are trying to paste external datas into bCNC at startup. Is this the case ?
If you are copying malformed external datas, then you will have a failure. 

I think that we are far from the nominal case.
We could catch the exception as described above.

DodoLaSaumure, 2020-01-09:
Does the error message remains after this commit a377045 ? @onekk 
Thanks forwards
Have a good day
Pierre

Harvie, 2020-01-09:
> I understand that you are trying to paste external datas into bCNC at startup. Is this the case ?
> If you are copying malformed external datas, then you will have a failure.

I think if user pastes non-json plaintext from clipboard, it should be treated as g-code and pasted as g-code. same way as if you import g-code file. maybe we can just reuse the code that we use to save and load g-code files to bCNC instead of json...

onekk, 2020-01-09:
The problem is when you hit the "paste button" at startup, you could have some "not pertinent" data on it, developing is quite usual to have some codelines in the clipboard generated by copying an paste code and when you test the program, if you hit by mistake the Paste button that happens to be near the load button in the editor window, I make this "mistake" sometimes when trying to load a new file, the problem is the robustness of the program, if you copy and paste proper data all is going well.

Regards

Carlo D.

DodoLaSaumure, 2020-01-09:
Hi @onekk @Harvie 
Witth this last commit 7801ab2 I am able to copy paste gcode from external apps. Does it work for you too? 
Thanks forwards
Pierre

DodoLaSaumure, 2020-01-09:
I think 8ea2755 is more flexible