vlachoudis/bCNC #1338: Tool change: avoid extra probing of first tool

auhopu, 2020-01-09:
When using "Manual Tool Change (WCS)" or "Manual Tool Change (TLO)" Z-zeroing and calibration must occur first.

Typically both these actions can  be done with the first tool. Then when the program starts the first M6 will trigger a probe for the first tool once more, which will land at the same Z as the calibration run that already took place. This is a waste of time.

To overcome this I guess one could manually delete the first M6. Wouldn't it be better if there was an option in bCNC which when ticked would

- either ignore the first M6 and still rely in (pre)calibration
- or run the calibration upon the first M6

I wonder whether more people would find that useful.

gauna85, 2020-03-01:
Or just an additional option to raise a question box if tool change should be performed or skiped

MARIOBASZ, 2020-03-01:
Maybe this helps you:
I think I remember that in bCNC if you call M6 and the new tool number is the same as the current one, you don't make any moves. Unfortunately, you cannot answer if the problem is grbl or bcnc, you cannot choose the current tool number.