vlachoudis/bCNC #1341: a stable release?

greyltc, 2020-01-20:
Any chance you could tag a commit as being stable? That makes packaging easier. It's been a few years since 0.9.11 was tagged.

Harvie, 2020-04-09:
i added 0.9.14.304 which is not exactly a stable release, but that's what i currently use...
maybe we can find some older 0.9.14 release to tag as well. until 0.9.15 is ready... If you know some nicely working commit ID i can tag it.

BTW have you considered building packages based on pypi releases? https://pypi.org/project/bCNC/

What distro are you packaging for?

Harvie, 2021-02-23:
@greyltc what do you think about making packages based on pip releases? is that good solution for you?

greyltc, 2021-02-23:
Sorry, I missed your earlier comment. I'm packaging for Arch.

For many reasons, when I package stuff I strive to fetch the source tarball I build from from as close to the authors as I possibly can. Would be nice if the tags were in this repo instead of somewhere on pypi.org since this is where I plan to get the source to build from.

Harvie, 2021-02-23:
> I'm packaging for Arch.

Nice to hear that. I use Arch almost exclusively, especialy during bCNC development. I think we can figure something out.

BTW pip/pypi can be used inside PKGBUILD to fetch the package:
https://wiki.archlinux.org/index.php/Python_package_guidelines#pip
https://wiki.archlinux.org/index.php/Python_package_guidelines#Source

Harvie, 2021-02-23:
Also i think it should be easy to add some kind of rolling tag called "stable" that would always point to the latest commit uploaded to pypi. However lately we've managed to keep git master more or less stable anyway...

greyltc, 2021-02-23:
The reason I prefer to grab the source from here is that way I can be sure that the content of the source tarball I get matches what people have been reviewing here.

Do you think there's some way we can make it okay to build the project from the source tarball snapshots of the repo as hosted on github.com? I'm happy to add you as a co-maintainer to the AUR package if you'd like btw! https://aur.archlinux.org/packages/bcnc/

Harvie, 2021-02-23:
There are these two files that i currently use for releasing:

https://github.com/vlachoudis/bCNC/blob/master/setup.py
https://github.com/vlachoudis/bCNC/blob/master/makefile

Whenever i feel like there were enough changes in bCNC to be useful for people and bCNC is reasonably stable, i simply increase the version number in `setup.py` and commit it (usualy with message being "pypi bump" or something). Then i call `make upload` which automaticaly uploads current commit to pypi.

greyltc, 2021-02-23:
Got it. I'll try to say on top of ensuring the stable Arch package matches the commit where you bumped the version number in `setup.py` then.

Harvie, 2021-02-23:
I've modified the makefile so it does tag latest commit uploaded to pypi with tag called "pypi" each time i do the upload.
So you can perhaps use this.

Harvie, 2021-02-23:
Is such approach good for your use case?

greyltc, 2021-02-24:
Yep, that should work. Thanks!

MicroJoe, 2021-02-24:
Hi, I would like to package bCNC for Debian. We usually use the [tags page](https://github.com/vlachoudis/bCNC/tags) with `uscan` to detect newer versions. However, latest version listed here is 0.9.14.304. Is it possible to tag every new release with proper tags instead of using the "pypi" tag? Because with this "pypi" tag we cannot detect when a new version is released, and what is the version number.

You would need to push a new tag each time the version is changed in `setup.py`. Does this seems reasonable to you?

Harvie, 2021-02-25:
It is possible. However i am bit afraid that there will be too much tags and it will get confusing here on github...

Harvie, 2021-02-25:
On the other hand... We have Travis CI enabled.
Idealy i would like to use Travis to generate and upload packages for all platforms automaticaly for each tag (or even commit).
If you can provide custom Travis scripts, i can add them to our pipeline.

MicroJoe, 2021-02-25:
> It is possible. However i am bit afraid that there will be too much tags and it will get confusing here on github...

It is okay to have a lot of releases. See [svgelements](https://github.com/meerk40t/svgelements/tags) for example.

> Idealy i would like to use Travis to generate and upload packages for all platforms automaticaly for each tag (or even commit).
If you can provide custom Travis scripts, i can add them to our pipeline.

The Debian way of working is that packages should be manually imported, not automatically from upstream. However, if you are pushing tags, this is very easy to do using something like 3 commands to update to the latest version for Debian. I am currently pushing my patches as pull requests to the project so that both upstream and Debian are very similar in future releases.

greyltc, 2021-02-25:
Unfortunately, pre-built external binaries (like would come out of your Travis CI) aren't so useful for the Arch package. Our packaging system needs to fetch and build the source.

I'm also unsure why having a lot of tags in this repo is a bad thing. It honestly seems much more confusing the way it is now, with a bunch of tags in pypi.org: https://pypi.org/project/bCNC/#history that don't match tags here!

Harvie, 2021-02-25:
BTW i have weird conflicts between python packages installed using pip and ones installed by distro-specific package manager (pacman on ArchLinux). I kinda hate everybody for pushing their solution and not being able to agree on one package manager to rule them all. But i know this will probably never happen and might not be a good idea anyway.

It is kinda bold of distro maintainers to repackage python packages.
And it is even bolder of Python (and other langs) developers to install their own package manager to someone's distro.

I understand that python can't be bothered to handle all existing package managers and their edge cases.
I wish there was some seamless solution to integrate these ecosystems together.

greyltc, 2021-02-25:
Yeah, I feel you. I generally keep pip off my systems (absolutely never running it as root), except in exceptional circumstances when I'm forced to use it for some reason, I only use it with a virtual environment.

This way pacman is the only thing managing files outside my home dir on my system (as it should be imop).

Harvie, 2021-02-25:
Well, from the point of view of bCNC developer i prefer everyone using pip to install bCNC, because that is only thing i can directly support. I don't have time to maintain all distro specific packages.

Also if there is github issue and i fix something i put it on the pypi, so even beginners can easily upgrade it and test the fix immediately without having to wait for maintainer of his distro. (well not everyone has AUR with git packages). As long as we encourage new users to install bCNC using pip, they all already know how to do it and can upgrade properly if needed. Before that people were runing distro specific releases of versions several years old and reporting bugs.

> It is okay to have a lot of releases. See svgelements for example.

Currently i tend to release on pypi a lot. I would even say that i do pypi release after every commit/PR. Lots of these releases are just experimental bugfixes, that i need to get ASAP to the testers. Having them as tags might make people think that all of these old tags should be considered stable.

While i think most of the pypi releases are reasonable. There was no actual tested and "stable" release for years. I plan to do 0.9.15 in few days. 0.9.14.xx releases are more of development prereleases leading to 0.9.15. So i don't really don't want to draw much attention to them.

greyltc, 2021-02-25:
I totally understand that you can only target pypi and can't be bothered with every little distro's specific packaging idiosyncrasies. I'm willing to take whatever you have and try to do whatever work might be needed to get it packaged the way I think it should be for Arch and friends distros.

What if the tags you used on github were just for milestone releases? and your dev releases stayed as they are on pypi?
For example, if version numbers you use are of the format: X.Y.Z. whenever you bump X or Y, maybe you could place that tag in this repo. I'm sure such a thing is scriptable for your CI.

Harvie, 2021-02-25:
> What if the tags you used on github were just for milestone releases?

But that is what i currently do :-) I joined bCNC after 0.9.14 release and now we are approaching 0.9.15 tag.

Harvie, 2021-02-25:
On the other hand i am sure there will be lots of 0.9.15.xx bugfix releases worthy of inclusion to distros, because there are lots of open issues on github and i don't even have time to go through all of them (or at least close the solved and invalid ones).

Harvie, 2021-02-25:
I think it would make sense to tag something eg. ~once a month after the 0.9.15 release. There is little point to produce 6 linux packages a day...

Harvie, 2021-03-29:
@greyltc As for reply to your recently created unnecessary duplicate issue. This is where i've suggested the solution and you agreed on that:

[image]

If that is not enough for you, please see my previous comment:
https://github.com/vlachoudis/bCNC/issues/1341#issuecomment-785872882

greyltc, 2021-03-29:
The reason why you'd like to reuse the "pypi" string rather than a version number string in the tag name is to prevent the github tags page from getting cluttered with many tags?