vlachoudis/bCNC #1343: Problem with engraving and autoleveling

IZ7QEB, 2020-01-25:
Hi all!
I've a problem with my 3018 CNC and bCNC 0.9.14.30. 
I'm trying to engraving my circuit on PCB but I've some problem!
I've readed also the other issue but I can't found solution.
After make the circuit and exported the gerbers files, I'm obtaining the g-code with flatcam.
I'm trying to setting the parameters as follow (see pictures):
Flatcam 01
[image]
Flatcam 02
[image]
Flatcam 03
[image]
bCNC code
[image]

Importing the g-code in bCNC, all is ok, I see everythink on code. 
I set the X, Y, at start point. Then, I set Z (to 0) about 1 cm above the PCB for safe moving.
Then, I go in  probe menu, autolevel page, set the Margins, I look that the axis min and max are right; start the scan. It is done with the contact wire.
After finish, I see the little correction; the probe goes to start point. 
I click on ZERO and I look change the colors. 
Then, I click on start and the probe start the engraving. Now is the problem: more time, the v-bit is toucing the surface but is not engraving. 
Other time, when I set the Z to 0 at 0.01 cm below the PCB, I broke the v-bit .. Now I destroy about 4..5 v-bit.. 
What I mistaking to set ?
PLEASE.. HELP!
See the attachment..
[image]
[image]

MARIOBASZ, 2020-01-25:
If I remember correctly, where you declare z = 0, check that the depth is z = 0

IZ7QEB, 2020-01-26:
Hi MARIOBASZ, thank you for answer! 
Probably I lost some step or I'm wronging somethink.. But what ?
Surely, somethink in setting.. 
Any suggest? 
Thanks..

MARIOBASZ, 2020-01-27:
I think the problem is this: check that where you set Z = 0, the height is indeed Z = 0, otherwise it will sink excessively or float above the matetial.
I think my routine is like this:
1- I choose an arbitrary point that is going to be Z = 0 (in my case I usually choose x0 and 0, it can be any)
2- Zprobe, and I set there as Z0
3- autolevell
4- I go to the chosen point and set it as z0 (now I have the question of whether this point is needed or I can ignore it)
5- Milling of pcb
6 -...
7 -...
8- Welding: Refrain from copying my procedure: I have an innate ability to generate short circuits: it is one of the reasons why I am trying to mill pcb with two burrs (example 0.7mm and 0.10mm), for the moment interrupted, and an ATC (automatic tool changer) economic, luckily so far it seems in a good way.

kfmut, 2020-02-04:
@MARIOBASZ

> 4- I go to the chosen point and set it as z0 (now I have the question of whether this point is needed or I can ignore it)

I think it's good practice to choose point where you could easily re-probe for z0 if something goes badly or you are not using ground plane and there need for tool change. For some reason I'm having better results with repeative probing than with tool calibration and automatic tool changing, even managed to remove solder mask on pads without digging into copper.

MARIOBASZ, 2020-02-04:
I copied the method that Wegstr uses to remove the solder mask from the pad, adding a spring to the bit: I've only done one test, but the result is excellent

Harvie, 2020-02-04:
> adding a spring to the bit

Where did you get the springy bit? Have you made it yourself?

MARIOBASZ, 2020-02-04:
If I remember correctly I did it like this:
1-Insert the V Bit into the collet
2-I put the cap that comes on the VBit on the back of the VBit, to prevent it from falling
3-A spring (I think it was disassembling a pen, or one of 5mm in diameter) embedded in the cap
4- Place the nut with the collet on the spindle, which allows the movement of the VBit
I think it was before changing the 500W spindle, the internal height of the servo hole to fulfill its function against the end of the spring.
Now that I think about it, maybe with Eva rubber you can

MARIOBASZ, 2020-02-04:
[image]

kfmut, 2020-02-04:
@MARIOBASZ interesting approach, I will definitely check that! Thank you!

IZ7QEB, 2020-03-22:
Thanks for comments and suggest. But I have, always, this problem for autoleveling and engraving. 
Is there someone that help me, step by step? I'm italian, I speak also english. 
I'm using bCNC 0.9.14-dev (last downloaded)
Thank you!

sle118, 2020-08-10:
I've had similar problems in the past, and they were caused by a miss configuration that caused the Z axis stepper motor to skip upwards steps. The result is that the controller's Z axis tracking becomes out of sync with the physical position.

You could try slowing down the Z axis motion, as well as reducing acceleration.

roor, 2021-06-18:
I have used autoleveler and as you can see on screenshot:
z height should be -0.1434 but in fact it is -0.276
and it looks like it in real life (engraving to deep, should be -0.005 before autoleveling)
what is wrong?

i have set z before starting autoleveling and starting cycle, so z it properly configured 
[image]

kfmut, 2021-06-18:
@roor 

> what is wrong?

Nothing wrong here:

Z height from autolevel is between -0.107 and -0.167, let's say it's something like -0.13mm. Z-height from g-code is -0.143. Result is -0.273mm.

kfmut, 2021-06-18:
@roor

Looks like you have applied autolevel to g-code file and didn't delete probe information. Right?

roor, 2021-06-18:
@kfmut exactly
Just was going to write the same, so proper actions should be
* either apply it to gcode, and clean probe data ?
or 
* don't apply it to gcode, and just run cycle with with probe data left on screen?

kfmut, 2021-06-18:
@roor

> don't apply it to gcode, and just run cycle with with probe data left on screen?

Works for me, just don't delete probe info between different files(tracks, copper etc)

roor, 2021-06-18:
@kfmut good to know

thnx a lot for help, will try it tomorrow