vlachoudis/bCNC #1344: Calibrate shortcut

auhopu, 2020-01-27:
I would like to use a shortcut to be able to trigger a calibration. That way I could assign it to a keystroke on a gamepad (much easier than fiddling with a mouse). It looks like there is command available for the MDI (as ABOUT, CLEAR, HOME etc). Would that be an easy addition?

As an alternative I also tried a custom button macro, based on calibrate function in caliProbePage.py 

G53 G0 Z-2
G53 G0 X[toolprobex] y[toolprobey]
G53 G0 Z[toolprobez]
G91
G38.2 Z-50 F120
G1 F500 Z1
G38.2 Z-5 F30
G4 P1
%wait
%global toolheight; toolheight=wz
%update toolheight
G53 G0 Z-2
G90

...but it seems that even though I can have multiple gcode lines in a macro or a % line in a macro, I cannot have multiple gcode lines AND % lines.

auhopu, 2020-01-27:
Maybe a
```
elif cmd == "CALIBRATE":
    self.calibrate()
```
in Sender.py under executeCommand ?

auhopu, 2020-01-27:
Replying to myself :D

I'm  not sure whether this is the proper way, but adding this in the execute function of `__main__.py` worked:

```
                elif cmd=="CALIBRATE":
                        Page.frames["Probe:Tool"].calibrate()
```

I can run CALIBRATE from the command line, from a macro button and within a shortcut.

MARIOBASZ, 2020-01-28:
I am following your progress, it can be useful!

auhopu, 2020-01-28:
Ok, let me explain the reason behind this.

I run bCNC on a RPi and use a wireless Locitech keyboard. It has an integrated touch pad to be used as a mouse, which makes it compact and convenient. However, the touchpad is very sensitive and the beginning of a drag gesture -if touched fast/hard- can be misinterpreted as a tap, i.e. a left click. Hell can break loose if this tap happens to be on the wrong interface button at the wrong time.

For this reason I heavily rely on shortcuts that I have mapped to keystrokes on a gamepad. These buttons are more tactile, cannot trigger accidentally and can be pressed without even looking (the same way you don't look at your fingers when playing a game).

Up till now I had a dedicated button for probing. The button was mapped to f7 and the f7 shortcut contained the probing macro.

I recently got a tool length sensor. Now instead of re-zeroing with every tool change, I can zero & calibrate once and let all subsequent tool changes probe automatically with the tool setter. This is a huge time saver.

Since the touch plate I use for zeroing is NO and the tool setter is NC, I need to use the former in the beginning of a setup, then switch to the latter and start running the job.

So ideally, I would like to have a single button to:

(run the probe macro)
%msg PLUG TOOL SETTER
M0
(run the calibrate macro)

I have managed to assign all of the above to a macro button and it woks fine, except it is not shortcut-able (I have to use the touchpad).

I have managed to assign it in two halfs to two shortcuts, but not the whole thing. I press one physical button for probing, switch sensors, press another button for calibrate... then run the job. It seems I ran into another limitation there. A shortcut can contain a gcode macro, a command (CALIBRATE in this case), but not both at the same time.