vlachoudis/bCNC #1346: Definition CNC.vars["cutfeed"]

jophut, 2020-01-30:
Hi there,

OS: Windows 10
Machine: Proxxon MF70 with GRBL
I'm using bCNC 0.9.14-dev with the default "TEXT"-plugin via CAM-> Generator -> Text.
Now I'd like to set the cutting feedrate, but can't figure out where to set it.
I've found in the sourcecode that it is set from CNC.vars["cutfeed"], and it has a default value of 1200.

Harvie, 2020-01-30:
This just means the settings of globaly selected material are used. The same will happen in other tools, when you leave the feedrate empty. This is where you set it:

[image]

closed 2020-01-30 by Harvie

jophut, 2020-01-30:
Ah thnx!

And just for anyone who's also having issues with this:
1) Select the "Material" type in the Stock
[image]

2) Set the feedrates in the material
[image]