vlachoudis/bCNC #1347: bCNC stop when made the first block

gnurc, 2020-01-30:
Hello,

I'm newbie with a great software and I'm trying to mill a fr4 panel but when you make the first block, bCNC stays in IDLE state. I have tried the same gcode with another gcode sender and it doesn't stop, what could it be due to?

https://imgur.com/a/y7jFpwK

Best regards

Harvie, 2020-01-30:
Software bugs can happen, but usualy it's g-code error or hardware error (eg. bad usb cable, EM interferrence caused by spindle and unshielded wiring,...).

MARIOBASZ, 2020-01-30:
Is there an error code message?

gnurc, 2020-01-31:
Nope, not error code. I think that bCNC is waiting for some command.

If I changed in Probe--> Tool--> Policy to "Ignore M6 commands" or "Manual (No Probe)" works.
[image]

I was using WCS that is the recommend in the wiki page. 

I prefer to use manual tool changed with probe if it's possible.

DeanRM, 2020-03-20:
Check the bottom of #1371 for a possible answer to your question.

dantiel, 2020-04-15:
Hey folks, i am having the same issue, bCNC 9.14-dev stops after almost finishing a block. i think the problem is the generated "(exiting)", after whose it doesnt go out of IDLE state. please help.

also: what is slowest feedrate i can use?

MARIOBASZ, 2020-04-15:
electrical noise?
on the terminal screen, do you report any error code?

MARIOBASZ, 2020-04-17:
I disable hard limit switches, I only use them for homming and soft limit switches

dantiel, 2020-04-17:
i have not enabled limits. i dont know how to work around this, the console just says that it went into IDLE mode, but pressing pause twice doesnt change anything... only thing i can do is stop the task, and try from start, but it usually hangs at the same spot

dantiel, 2020-04-24:
i dont know if this is already fixed, but after updating my raspbian distro, i found some other weird random behavior appearing. When i click to "Run" the gcodes it is displaying -9999, machine moves few of milimeter, and job ends. sometimes however the software is in finished job mode but the job cintinues in the background, this time without being interrupted by the other bug described in this thread. it may be the same underlying issue however, that could have to do with some weird behavior of the python drivers?