vlachoudis/bCNC #1348: Milling the center of the outline

gnurc, 2020-01-31:
Hello,

When I run the job the spindle is milling on the edge, How can I make it go through the center? As if it were a single line.

https://imgur.com/a/6vYWtix

Best regards

gnurc, 2020-01-31:
Hi again,

I changed the cut bit diameter in pcb2gcode to 2mm and now look better.

[image]

 But if I executed _profile out_ and _cut_ 

[image]

Is it possible to cut it and make it look like in the first image?

MARIOBASZ, 2020-01-31:
Why do you want to do that? I guess choosing a thin track width in pcbgcode generates the path you want

gnurc, 2020-01-31:
Curiosity, I thought it was better but I tried and cut well in one pass. You are right is enough so.