vlachoudis/bCNC #1349: bCNC freezes on pocket action

AlfredoCubitos, 2020-01-31:
Hi,

after clicking the Pocket button twice bCNC freezes with this error:

```
self.canvas.winfo_id(): 100663381
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/usr/lib/python3.7/site-packages/bCNC/ToolsPage.py", line 1747, in execute
    self.tools.getActive().execute(self.app)
  File "/usr/lib/python3.7/site-packages/bCNC/ToolsPage.py", line 909, in execute
    app.pocket(name)
  File "/usr/lib/python3.7/site-packages/bCNC/__main__.py", line 1827, in pocket
    msg = self.gcode.pocket(blocks, diameter, stepover, name)
  File "/usr/lib/python3.7/site-packages/bCNC/CNC.py", line 4011, in pocket
    newpath.extend(self._pocket(path, -D*diameter, stepover, 0))
TypeError: 'NoneType' object is not iterable
```
steps to reproduce:
1. select object
2. click Pocket -> everything ok, showing the path
3. click Pocket again -> freeze :-(

MARIOBASZ, 2020-01-31:
if the bit diameter does not have space inside the silhouette to generate pocket, that serious error occurs

gnurc, 2020-02-01:
Ok, Thank you.

Harvie, 2021-02-23:
There is new pocketing tool called "offset" in latest bCNC which is not affected by this crash, old profile/pocket tools will be removed in future release.
Please test it and we will reopen the issue if the problem persits.

closed 2021-02-23 by Harvie