vlachoudis/bCNC #1350: CAM->Generator->Text

GHswitt, 2020-01-31:
Version: bCNC 0.9.14.301

When trying to use CAM->Generator->Text with different ttf font files I get:

Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/usr/lib/python3.8/site-packages/bCNC/lib/tkExtra.py", line 866, in handleKey
    active = unicode(self.get(ACTIVE))
NameError: name 'unicode' is not defined

and

Text abort. ... Can't read this font file!

DodoLaSaumure, 2020-03-06:
cbd6162 tested with both py2 and py3, should solve your problem
Any feedback appreciated
Best regards
Pierre

DodoLaSaumure, 2020-03-06:
82c27d3 tested with more ttf fonts.
Can you please test and give your feedback.
Thanks forwards
BEst regards 
Pierre

GHswitt, 2020-03-25:
Works fine for me, thank you very much
Best regards
Sebastian

DodoLaSaumure, 2020-03-26:
Could you please reopen this issue, until merged on the head of the project?

DodoLaSaumure, 2020-03-26:
Thank you @GHswitt and thank you for your feedback.
Have fun with ttf fonts
Pierre

closed 2020-12-28 by Harvie

Harvie, 2020-12-28:
Reportedly fixed in this PR. Please test and close this issue if everything is OK:
https://github.com/vlachoudis/bCNC/pull/1405#issuecomment-751580531