vlachoudis/bCNC #1351: Y axis compensation

gnurc, 2020-02-02:
Hello,

It's possible to compensate the Y axis with bCNC software?

My X axis is perfect but the Y axis has 1,5mm in 300mm.

I have found this but it's ready implemented in bCNC 0.9.14_dev?
https://github.com/vlachoudis/bCNC/issues/700#issuecomment-357671007

Best regards

MARIOBASZ, 2020-02-02:
Orthogonality of the machine and backlash, in my opinion, the ideal would be from grbl.
Perhaps bcnc can be compensated in the sending of the coordinates; Another way would be a plugin that modifies the original gcode.
Mechanically, hardware: If the "X" axis is the one that runs on the "Y" axis, this axis is the one that is different from 90 degrees.
Can you rotate it?
I have found a quick method for it:
1- cut a rectangle
2- flip the inner rectangle to match one of the "Y" sides of the outer rectangle
3- With the web cam it is more precise, if not, the ortiginal segment is from the point of intersection of the two squares on "Y" until halfway between the two rectangles on the "X" axis
4- There is also an economical way to control perpendicularity of the spindle using a 3mm wire in the form of "N" and using z probe instead of pair of comparator clocks (expensive).
By software, it occurs to me that you can compensate using scaling with different values ​​for X and Y.
My brain does not reason well on Sundays at this time, I do not know if it is a suitable idea, and if it is (it should be: if a square draws it as a rhombus, it should be scalable so that it effectively draws a square).
Comment if it solves the problem, although it is not the best solution (I think that scale converts arcs into segments, does not return M3 M4)

MARIOBASZ, 2020-02-02:
Maybe use rotate?

gnurc, 2020-02-03:
Thanks Mario,

Yes, would be from grl.

I'll try it.

Best regards

closed 2020-04-16 by gnurc