vlachoudis/bCNC #1352: Severe issue for macros since last update

BernardG, 2020-02-03:
I am on Ubuntu 19.10 with Python 3.8

% on a line trigger this error
```

File "/usr/local/lib/python3.8/dist-packages/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/usr/local/lib/python3.8/dist-packages/bCNC/Utils.py", line 798, in ok
    config.set("Buttons", "command.%d"%(n), self.command.get("1.0",END).strip())
  File "/usr/lib/python3.8/configparser.py", line 1201, in set
    super().set(section, option, value)
  File "/usr/lib/python3.8/configparser.py", line 894, in set
    value = self._interpolation.before_set(self, section, option,
  File "/usr/lib/python3.8/configparser.py", line 402, in before_set
    raise ValueError("invalid interpolation syntax in %r at "
ValueError: invalid interpolation syntax in '%_mydia=app.entry("Entrez le diametre:");
```

It does the same on Windows 10, with Python 3.8 too.

Obviously, without the "%", nothing works!
Looks strange to me. Have macros been tested with Python 3.x?

Beside this, everythings seems to work OK

GitHubCaps, 2020-03-02:
In Python3, there are a lot of changes to how the configuration file is read. You now have to escape percent signs with another percent sign, so `%` is now `%%`. However, there are more issues. `basestring` is now just `str`. If you are comfortable editing your files, I have a proposed fix that you can try. I am not ready to add a commit for changes yet, until I have worked out some of the other issues.

Going off off one of your earlier post, you can place this in your `bCNC.ini` file in your home directory, not the one in `bCNC` main directory.

```
name.8 = Sonde XYZ
icon.8 = 
tooltip.8 = 
command.8 = %%_mydia=app.entry("Entrez le diametre")
	%%_demidia=float(_mydia)/2
	%%_epZ = float(0.1)
	%%_epX = float(0.1)
	%%_epY = float(0.1)
	%%_correctZ = str(_epZ)
	%%_correctX = str(_demidia + _epX)
	%%_correctY = str(_demidia + _epY)
	%%msg Debut de Sondage
	G92.1             ("Suppression de toutes les valeurs d'offset existantes (92.2 non pris en charge, utilisez 92.1")
	G38.2 Z-35 F100   ("Sonde en Z de 45 mm maximum à 50 mm/min")
	G92 Z[_correctZ]  ("store la valeur trouvée + 0.1 mm-essai feuille alu")
	G91 G0 Z3         ("remonte à 3mm rapidement")
	G91 G0 X-25       ("Recule rapidement en X de 50 mm")
	G91 G0 Z-5        ("descend à Z-5 rapidement, en préparation du sondage en X")
	G38.2 X25 F100    ("sonde en X de 50mm - On doit forcemment trouver, puisque c'est de là qu'on vient")
	G92 X-[_correctX] ("store la valeur trouvéé + correction")
	G91 G0 X-3        ("recule de 3mm en Z")
	G91 G0 Y-25       ("Recule rapidement en Y de 50 mm")
	G91 G0 X25
	G38.2 Y25 F100    ("sonde en y de 50mm - On doit forcemment trouver, puisque c'est de là qu'on vient")
	G92 Y-[_correctY] ("store la valeur trouvée + correction")
	G91 G0 Y-3        ("recule de 3mm en Y")
	G91 G0 Z20        ("remonte de 10mm")
	%%wait
	%%msg Sondage Terminé
	%%app.setStatus("Vous pouvez cliquer sur le cercle dans Control pour vérifier")
```
Note that you can use accents now and notice the double `%%`.
In `bCNC`, when you edit, it will show single and you must add the extra one back in. However, that should not be a problem if you do not need to edit your configuration. It may even be easier to edit your .ini file before launching `bCNC`

At the top of bCNC.py in the main directory, you need to add a fix for `basestring` as follows.

Replace this
```
try:
	import Tkinter
	from Queue import *
	from Tkinter import *
	import tkMessageBox
except ImportError:
	import tkinter
	from queue import *
	from tkinter import *
	import tkinter.messagebox as tkMessageBox
```

with this
```
try:
	import Tkinter
	from Queue import *
	from Tkinter import *
	import tkMessageBox
	basestring = basestring
except ImportError:
	import tkinter
	from queue import *
	from tkinter import *
	import tkinter.messagebox as tkMessageBox
	basestring = str
```

You should see this in `User configurable button`
[image]

To have a better view of the command list, I edited `Utils.py` line:761 and changed this
```
		self.command = Text(self, background=tkExtra.GLOBAL_CONTROL_BACKGROUND, width=40, height=10)
```
to
```
		self.command = Text(self, background=tkExtra.GLOBAL_CONTROL_BACKGROUND, width=120, height=30)
```

Hopefully, that will work for you until this problem gets resolved, but according to the python docs, there will be even more changes when 3.9 comes out!

Please let me know your results if you make these changes.

BernardG, 2020-03-08:
I am out of town now, but I am going to test this once I am back home, toward the end of the week. Thanks for taking the time to answer to me on this issue.
Best regards,

Bernard

 

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 

 
 
 
 
 
 
 
 
 
 

 
 
 
 
 
 
 
 
 
 
 
 

 <https://user-images.githubusercontent.com/38666078/75718939-76235480-5c99-11ea-9e39-df43ba58b6ea.png>

BernardG, 2020-03-22:
Hi @GitHubCaps 

Sorry for not answering earlier! I finally got the time to get back to this. I implemented your different changes in the code, but it look like I have a path issue: changes in bCNC.ini in my home directory do not appear in the command button.

GitHubCaps, 2020-03-23:
/home/username is the location, but it may be hidden since it is a dot file. Ctrl+H will unhide the files in the directory. The actual name is .bCNC.ini as in /home/{user}/.bCNC.ini hope that helps.

BernardG, 2020-03-23:
Did not notice the DOT in front of bCNC! Argh.. Thanks….

GitHubCaps, 2020-03-23:
I apologize for the confusion, my fault for adding the .ini extension where it doesn't belong!

BernardG, 2020-03-23:
No worries! Know I have something to play with... Before, i was unable to see the text of the macro, as it was not loading, because of the french apostrophe in the comments, like here:
"On doit forcemment trouver, puisque c'est de là qu'on vient"
Seems it was not a problem before, but it's an easy one to fix: Put my comments in English instead... :-)

once I have checked it's actually working as expected, I am going to create more probing macros.

I would like to be able to have those tests made before closing the subject, if it's ok?

GitHubCaps, 2020-03-23:
Not a problem. I would appreciate your thoughts on this as well before I go committing changes to the main branch. I have worked out a ton of bugs on my end, but would rather do more testing first.

GitHubCaps, 2020-03-23:
If you have issues, feel free to post your macros so that I test on my end to hopefully get it resolved for everyone.

BernardG, 2020-03-23:
I have a first error "basestring is not defined". I have modified bCNC.py as you recommended:

[image]

After that, I have an error on basically all variables names. The macro itself is the same that you already have.

GitHubCaps, 2020-03-23:
Comment out the two basestring lines that were added. Put a separate try/except block just above. 

```
try:
    basestring
except:
    basestring = str
```
I am my mobile right now, so can not do a proper edit. If you copying the code, make sure to replace the 4 spaces with an actual tab. If you're not sure, go lower in the code in the pyfile and copy and paste an actual tab in it's place. I hope that makes sense.

GitHubCaps, 2020-03-23:
That last basestring = str
can also be:

basestring = (str, bytes)

May not be necessary, but wouldn't hurt either

GitHubCaps, 2020-03-23:
I made another edit up above to reflect a better method to try. NameError may help.

GitHubCaps, 2020-03-23:
Depending on what you are using as an interpreter, pylint for example, can raise errors and cause problems.

BernardG, 2020-03-23:
Same errors, even after implementing your suggested modifications. I know I have pylint in my path, I'll remove this folder from my path (I am on Ubuntu 18.04, if it can help)

GitHubCaps, 2020-03-23:
You don't have to do that, I just added another fix for pylint. It is a comment after the defined variable, see above. Thanks

BernardG, 2020-03-23:
Just saw this. Where should I put this?

GitHubCaps, 2020-03-23:
If you are asking about the try/except, just above or below where tkinter is being imported in its own try/except block. Just remove the basestring entries that you first put in place

BernardG, 2020-03-23:
No I was asking about # For Python 3, pylint: disable=redefined-builtin,invalid-name
Seems I have to put it like you did, as a comment?

GitHubCaps, 2020-03-23:
Yes, precisely

GitHubCaps, 2020-03-23:
Also be sure to check that your config file didn't get corrupted in any way. I forgot to mention that you should have made a copy beforehand, just figured you would have

BernardG, 2020-03-23:
Same error. Which config file? Sorry, did not do any backup before.

BernardG, 2020-03-23:
Going to stop for the day, will be back at it tomorrow. Thanks for helping!

GitHubCaps, 2020-03-23:
No problem, your config ~/.bCNC
You can launch from your bCNC with:
python3 -m bCNC -d
from your regular terminal to see if that helps. For more it's python3.8 in the commandline

GitHubCaps, 2020-03-23:
Just thought I should mention this, since you probably aren't worried about keeping backwards compatibility with python2, just get rid of that try/except (comment out if you want) and search for basestring. There is only one instance in the file and replace it with str. Just those three letters should solve your problem. Real pain dealing with these changes sometimes and as far as I can tell, it's only used for the entry box that the user inputs to, which of course is what you need for your macros to work.

BernardG, 2020-03-24:
I did that and tested; same error. I searched and found basetring somewhere else: line 3080 in tkExtra.py, but then it trigger another exception "TypeError: can only concatenate str (not "bytes") to str

I don't know enough about bCNC and Python to solve this.

GitHubCaps, 2020-03-24:
Just had to go back in the repo here to check why I don't have basestring in my tkExta.py file. Turns out that they put in a fix for issues #1314 #1310 that I don't have in my file. Well, it obviously is not the correct fix, I will do a diff again to see what changes there are from mine and find a better solution

GitHubCaps, 2020-03-24:
In tkExtra.py, comment out lines: 3079, 3080, 3081, 3082, 3085, 3086, 3087, 3088, 3089. Copy line 3085 and paste it under line 3084, then replace: `tmpstr =` with `return`. Be sure to comment it. I don't know why your interpreter is refusing to work with the redefined-builtin, but it is obviously not a solution that will work for everyone, good to know

GitHubCaps, 2020-03-24:
By the way, did it tell you in what file and line the `TypeError` occurred?

BernardG, 2020-03-24:
For the TypeError, I have a pretty long traceback :
[image]

I am going to test your modifications now.

BernardG, 2020-03-24:
OK, in fact, it's worst than I thought... (sorry!)

even after all thes changes, I still have the `basestring not found` error.

So I hard coded, in my macro, a value for the tool diameter, thinking if the problem is only with basestring, it should now work.

Well, no! I have an error for each of my variables, for example, I have now
```
%%_mydia = 6;
%%_demidia = float(_mydia/2);
```
I get "_mydia not found", and so on, for each variable.

BernardG, 2020-03-24:
My machine is connected to my Linux Box. Tomorrow I am going to connect it to my Windows computer, to see if it makes any difference.

GitHubCaps, 2020-03-24:
`My machine is connected to my Linux Box. Tomorrow I am going to connect it to my Windows computer, to see if it makes any difference.`

Unfortunately, no! Don't even waste your time trying on windows, the problem is with Python 3. Python 2 was very permissive in how it handled str, bytes, unicode, etc. Python 3 requires being very specific with your input. I will try to get a package together for you to test out and I will post it here for you to download and try.

GitHubCaps, 2020-03-24:
Thanks for showing me your log, know I know another place to look

BernardG, 2020-03-24:
OK, thank you for your time and your help. I'll test your package when you're ready.

GitHubCaps, 2020-03-26:
Here is a test-build for you to try. Unzip it where ever you like and open a terminal to launch e.g. `python -m bCNC -d`. It is self-contained, the userIni file is named `bCNC_userConfigFile.ini` in the same directory (your history file too) that you launch the terminal in, no need to go tracking it down. You can also delete it to start from scratch. However, I would like you to try it `as-is` for the moment. Please note that the terminal will be a bit noisy, because I have quite a few print statements to help out with debugging. Good luck, I hope this solves the issues this time!

[bCNC-test.zip](https://github.com/vlachoudis/bCNC/files/4384424/bCNC-test.zip)

On a side note, I added a feature (though not fully implemented) to help out using the user entry box.

%%_mydia=app.entry_num("Entrez le diametre", "your-title-here", input=3.0, type_="float", from_=None, to_=None, min_=1.0, max_=10.0)

`input=3.0` is a default setting to start off with (already avail. in the default entry box...)
`type_="float"` or `type_="int"` is to help make sure that it is valid input, no str (text vals) allowed.
`min_=1.0` and `max_=10.0` should be self explanatory, you supply them and if the user is out of range, then they are out of luck. Note, so far, it will continue with evaluation, just won't execute the code (may even have to restart the program to put new values) or at least it shouldn't.

Oh yeah, and like I said before, if you right click the user button to edit it, the `%%` turns back into `%`, so be sure and add it back. Not sure at the moment how to fix that, just know that it is annoying that they changed about as much as they could to make things nearly impossible to be backward/forward compatible!

GitHubCaps, 2020-03-26:
Darn, just realized that I forgot to delete the byte code files after testing on python 2.
` rm -r *.pyc` to the rescue!

BernardG, 2020-03-27:
sorry, I did not have the time to test yesterday. Just installed your zip, and I got the day before yesterday:

[image]

To be sure, I had installed apt-pkg and put it in my PATH environment variable. I'll check that again.
Maybe it's my Python which is having a problem?

GitHubCaps, 2020-03-27:
@BernardG 

[bCNC_runme.zip](https://github.com/vlachoudis/bCNC/files/4395817/bCNC_runme.zip)

Try this one instead. Go into the directory where you see the shell file `bCNC_runme` and launch it from terminal:

`./bCNC_runme`

is all you should need to do.

GitHubCaps, 2020-03-27:
Sorry about that, I was in a hurry a forgot that it may have been trying to load the legacy script, so put up a new file. I rename the old file to `bCNC_off` take a look in the shell script I mentioned above to configure if need be.

GitHubCaps, 2020-03-27:
Also, where you have:
`/home/bernard/Documents/bCNC-test/bCNC.py` showing, tells me you were not in the correct directory.
Should have been inside the `bCNC` directory where the files are...
`/home/bernard/Documents/bCNC-test/bCNC/bCNC.py`

BernardG, 2020-03-27:
I'll check that tomorrow. I changed the directory structure, so I was in the proper directory.

GitHubCaps, 2020-03-27:
No problem, just reluctant at the moment to put push any changes to the main branch until we can resolve the issues here. Hopefully, after you have successfully accomplished it on your end, then I will see about that, but possibly have others try to before then. I guess we need another branch like a nightly or weekly build.

GitHubCaps, 2020-03-28:
Just some info:
Don't know exactly why you didn't have `apt-get` (apt is your package installer) let alone why it prompted it you to install it. I am still using an older version of Linux Mint 18.3 Silvia (to be exact) to develop with. I can tell you that it didn't find `__main__`  from the package (`python bCNC` running from the commandline) and was prompting to install it. There are so many different ways that (in a linux environment) that things exist (or don't) and linux will prompt you to install a missing package if it doesn't find it (never use to be that way in the old days), that makes it hard to pinpoint what the problem may be,  however this is probably my fault for not being clear (ahem, my apology here) I guess that I expect that users know how to run things as I do. I do not install, nor do I run, from a pip install or the like, that makes it easier for others to install. I am more used to making it so, than the other way round. Anyways, glad you showed me, because that pointed to my own error, not yours. You shouldn't have to install anything if you can run bCNC in normal mode. Key point is that I should probably create a separate branch to make it easier for testing without interfering with the main repository here. I hope all that makes sense!

Best regards...

BernardG, 2020-03-28:
It works! It has been a bit of a mess for me to test, as yesterday I managed to damage my Linux box, as I removed Python 3.7 from Ubuntu, but in the same time it removed a lot of needed utilities, si it does not boot up anymore!

So I did my tests on Windows, where it ended up working perfectly. I am going to try another, more complex, macro, and I'll let you know.

BernardG, 2020-03-28:
OK, I am trying to write a more complex macro, but I get errors I don't understand, maybe because of Python 3.8 "pickiness"

this line:
`%%_distmvt = (float(0.5) * _flong) + 10.00`

[image]

give me this error:
[image]

I am attaching the whole ini file. I have this kind of operation in different places, and they (obviously! ) ALL give me this kind of exception.

[bCNC_userConfigFile.zip](https://github.com/vlachoudis/bCNC/files/4397106/bCNC_userConfigFile.zip)

GitHubCaps, 2020-03-28:
I also forgot to mention that you can just make an archive of your original install of bCNC, then delete the files there and put the new files in there place. That may be easier for you than messing around with paths in your environment. Or you can simple extract the new files over the old, up to you.

BernardG, 2020-03-28:
I believe I have to reinstall my Linux box, just not much time for this right now.

Any idea about how to write this `%%_distmvt = (float(0.5) * _flong) + 10.00` without triggering an exception? (I know it's more a Python question than a bCNC question)

Something else: I believe to have read somewhere I could embed python (with %%) inside a GCode file, obviously without the dialog boxes. Is that right? I would love to try this, as it would allow me to have Fusion 360 generate G-Code for probing, with a few (well, quite a lot, in fact!) modifications to GRBL post-processor

GitHubCaps, 2020-03-28:
Here you go! Works on my end now, notice all the changes I made to the syntax. Stop using the old form that works for Python2 and switch to how it has to be for Python3. You should notice the difference right away if you do a file comparison.

[bCNC_userConfigFile.ini.zip](https://github.com/vlachoudis/bCNC/files/4397440/bCNC_userConfigFile.ini.zip)

GitHubCaps, 2020-03-28:
> I could embed python (with %%) inside a GCode file

I haven't tried it inside a file myself, I have always stuck to the old school code.
I believe it depends on the controller and how it interprets the code, all you can do is try and find out. As they say, never hurts to try!

GitHubCaps, 2020-03-28:
One thing that may help you is an editor with syntax highlighting, if you don't already use one.
[image]
It helps me to spot mistakes faster for sure.

BernardG, 2020-03-28:
Thanks (a LOT!) for your help! Yes, I do use VS Code form Microsoft. Precious tool...

BernardG, 2020-03-28:
PS, like you say, never hurts to try, so I will, and I'll let you know how it goes.

GitHubCaps, 2020-03-28:
> I'll let you know how it goes.

Please do, curious myself now!

BernardG, 2020-03-28:
this `%%_distmvt = float((0.5 * _flong) + 10.0)` is still giving me the same error. What I don't get is why 0.5 is seen as a sequence.

GitHubCaps, 2020-03-28:
That's strange, I just ran it on mine all the way through. I will have to do some more checking, but it may not be until tomorrow, leaving town for the rest of the day.

BernardG, 2020-03-28:
[image]

BernardG, 2020-03-28:
It's OK, this can wait. It has been a long time already. And I do appreciate your help, very much.

GitHubCaps, 2020-03-28:
Maybe try removing my comments in the file to see if that helps

BernardG, 2020-03-28:
OK, will do. Have a nice week-end.

[20 more replies]

closed 2020-03-30 by BernardG