vlachoudis/bCNC #1353: Use probe data in button or block

MARIOBASZ, 2020-02-04:
Is there any way to get it to the point of the last probe?
I have tried to write
G53 G0 Z-40
G91 G38.2 Z-5 F100
G53 G0 Z [prbz]
but it goes to the last position informed according to $ # before these movements:
That is, it goes to an old probe point
Any way to update, to respond as when polling from the control panel, to perform it from a button or to test writing within a block in Gcode?

Harvie, 2020-02-04:
I am not sure... Does adding `%wait` help?

MARIOBASZ, 2020-02-04:
Hello @ Harvie, unfortunately not, even doing two probe the same, go to the last value before the macro.
I guess it's taking the value that grbl sends, but I don't know how to do it in the macro, just like in the probepage file

MARIOBASZ, 2020-02-05:
Grbl sends the position after probe. It must be simple: how do I update the values to cnc.vars, to update the probe data from the macro?

MARIOBASZ, 2020-02-05:
In button and macro it doesn’t work , but written inside a block and executed, it works correctly

truglodite, 2021-02-27:
Same problem is occurring for me as well. Macro names are not working as expected for probing with a macro button. Values used for prbx etc are values from before starting.