vlachoudis/bCNC #1354: Laser and solder paste

MARIOBASZ, 2020-02-05:
Anyone with a laser, could you test or tell me if your laser can weld solder paste or failing tin wire?
Maybe just an idea, but maybe a laser machine (possibly with more heat source) could be used to weld pcb boards.

Harvie, 2020-02-05:
There are lasers of various wavelengths and power specifications. Generaly speaking lasers range from not being able to cut black paper to being able to vaporize steel. So there is laser able to melt solder.

But personaly i don't think this would be too practical in hobby sphere for several reasons.

You need lots of power to melt shiny metalic material. laser absorption dependes on reflectivity of surface. tin is shiny, while components and PCB are not. so any slight misalignment or reflection will cause burning the components or board near the solder joint.

What problem will you solve by using laser to solder? You can just put whole board to heated chamber or on heated plate and it will reflow the paste.

But on the other side, this has already been done: https://www.youtube.com/results?search_query=laser+soldering

Harvie, 2020-02-05:
https://www.youtube.com/watch?v=nF2sVuXykCU

Harvie, 2020-02-05:
https://www.youtube.com/watch?v=KnyB9btlhS8

Note that they have temperature and visual feedback. They precisely aim the laser using computer vision camera and they also measure the temperature using IR camera. Otherwise the board will be burned.

Harvie, 2020-02-05:
https://www.youtube.com/watch?v=T2h7hagVfnA

MARIOBASZ, 2020-02-05:
OK thanks

closed 2020-02-05 by MARIOBASZ