vlachoudis/bCNC #1358: Where in bCNC can the version number be found?

jrlandau, 2020-02-21:
I'm sure I am missing something obvious.

Harvie, 2020-02-21:
Depends on what you need to know...

In the header of the window you have approximate relase version:

[image]

This is how you get pip package version:

```
$ pip show bCNC
Name: bCNC
Version: 0.9.14.303
Summary: Swiss army knife for all your CNC/g-code needs
Home-page: https://github.com/vlachoudis/bCNC
Author: Vasilis Vlachoudis
Author-email: vvlachoudis@gmail.com
License: GPLv2
Location: /usr/lib/python3.8/site-packages
Requires: numpy, Pillow, opencv-python, pyserial
```
If you run git version, do this in git repository:

```
$ git log -1
commit 6eddea4dc0adf68e26ea2963463069b95985cb41 (HEAD -> master, upstreamrw/master, origin/master, origin/HEAD)
Date:   Sun Dec 29 15:23:07 2019 +0100

    pypi bump
```

If you run something else than pip or git. Probably don't.

closed 2020-02-21 by Harvie

jrlandau, 2020-02-21:
Tomas,

Thanks.  I was so busy looking for a "help/about" that I didn't see what
was before my eyes.

One question:  In your email it looks like the example is of python3 (the
location field).  But I thought that bCNC does not support python3 yet.  Is
this so?

Joe

Harvie, 2020-02-21:
there is pip2 and pip3. pip is for whatever python version is default on your pc.