vlachoudis/bCNC #1359: Bcnc does not update tool number

MARIOBASZ, 2020-02-21:
The tool number is always 0, does not register change to T1, T2, etc. bCNC or grbl problem?

M4RC0660, 2020-03-13:
I've got the same problem

DeanRM, 2020-03-20:
Tool number is handled by your postprocessor at the same time the M6 command is issued. 
 
[image]

If I wanted to issue the tool number instead of tool name, I would use M06(<t>).

MARIOBASZ, 2020-03-20:
@ DeanMR, you are right, within the M6 code there is no incinent. However, my concern is as follows: If I start an operation, the spindle may be empty or with tool number n in place.
If bcnc can recognize this, it will be useful for my automatic tool changer project

DeanRM, 2020-03-21:
I suspect that bCNC has no code written to handle it.  The only place I see it is on the label during a tool change.  This situation is rare and I doubt any changes will be done.

MARIOBASZ, 2020-03-25:
I have checked the following:
1- Writing in command line (also in a button) tool number, this value remains persistent, it can be verified by sending $ G, the tool number is also updated in DRO.
2-Cannot set tool number from DRO Tool; it would be desirable to be able to set it like Feed or RPM
3-When executing a file, it always ends with T0. This is a feature of grbl, not bCNC.

GitHubCaps, 2020-05-16:
I believe, what is needed is updating the endmill database in bCNC to allow for toolNumber, toolOffset, spindleRPM, etc. so that variables can be pulled from there, just like commercial setups do.