vlachoudis/bCNC #1360: Problem updating bCNC on Raspberry Pi

jrlandau, 2020-02-21:
I've been using bCNC on the Protoneer hat, and now I am trying to update to the current version.  It appeared that I needed to upgrade python first, so I  upgraded to python 2.7.17.  This seems to work.  However, now I can't run bCNC; when I run "python -m bCNC" from the bcnc directory, I get ImportError: No module named tkinter. However, when I run "sudo apt-get install python-tk", I get "python-tk is already the newest version."

kfmut, 2020-03-07:
@jrlandau 

I've checked newer version of bCNC on my PI that wasn't upgraded for very long time(python's version is 2.7.9) and it starts OK. Maybe after python upgrade default python commmand now references to version 3(could be checked with "python --version")? Have you tried to lauch bCNC via bCNC.sh?

jrlandau, 2020-03-07:
@kfmut  Thanks for trying it.  I concluded that the problem probably is due to getting the 2 python versions mixed up somehow.  So I abandoned this approach and started again with a fresh copy of Raspbian Buster, which so far is working.  So I am closing this issue.

closed 2020-03-07 by jrlandau