vlachoudis/bCNC #1361: Call custom macro based on tool number during tool change?

vikhattangady, 2020-02-21:
_Full disclosure: I have not yet built or used a CNC machine. Just browsing features of different machines while I design my own machine in my head._
Inspired by E3D automated procedure for changing tool heads, ([something similar done using MPCNC](https://forum.v1engineering.com/t/automatic-tool-changer/10106)), was wondering about same using GRBL and bCNC. 

Some kind of custom script that reads the tool number, and chooses an appropriate macro. The macro will include the g-codes required to bring the z-carriage to the required position to drop/pick up the tool.

eg. M6 T1 - run macro "tool 1 pickup" (custom macro/gcode instructions); M6 T2 - run macro "Tool 2 pickup"; M6 T0 - manual tool change, run the existing tool change macro.

MARIOBASZ, 2020-02-21:
Yes, of course you can, I am advancing precisely in that, adding among the options (ignore M6, ....,) automatic tool change:
download current tool
grab new tool
compensate tool Z as in manual change.
Yesterday I have uploaded the problem that the tool data is not updated, only T0 appears, I do not know if it is a problem of bCNC, of grbl, or I am doing something wrong.

vikhattangady, 2020-02-21:
wow, thats great to hear. Really exciting times ahead for hobbyists like us.

closed 2020-02-21 by vikhattangady