vlachoudis/bCNC #1362: Python 3 issue

leventelist, 2020-02-24:
Dear Maintainer,

I tried bCNC with Python 3, and I get this:

python3 -m bCNC
new-config Utils <configparser.ConfigParser object at 0x7f6b38a38748>
Using fallback Utils.comports()!
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 98566234
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/lev/github/bCNC/bCNC/__main__.py", line 2716, in <module>
    main()
  File "/home/lev/github/bCNC/bCNC/__main__.py", line 2684, in main
    application = Application(tk)
  File "/home/lev/github/bCNC/bCNC/__main__.py", line 197, in __init__
    page = cls(self.ribbon, self)
  File "/home/lev/github/bCNC/bCNC/CNCRibbon.py", line 115, in __init__
    self.register()
  File "/home/lev/github/bCNC/bCNC/FilePage.py", line 395, in register
    (SerialFrame,))
  File "/home/lev/github/bCNC/bCNC/CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "/home/lev/github/bCNC/bCNC/FilePage.py", line 253, in __init__
    self.comportRefresh()
  File "/home/lev/github/bCNC/bCNC/FilePage.py", line 347, in comportRefresh
    for hw in i[2].split(' '):
AttributeError: 'NoneType' object has no attribute 'split'

Any clues?

leventelist, 2020-02-24:
The issue was that python3-serial wasn't installed. Sorry for the noise.

closed 2020-02-24 by leventelist

Harvie, 2020-02-24:
> The issue was that python3-serial wasn't installed. Sorry for the noise.

Weird. That should be installed automatically, because it's specified in setup.py:
https://github.com/vlachoudis/bCNC/blob/master/setup.py

leventelist, 2020-02-24:
Yeah... if you run setup.py. :-)

Harvie, 2020-02-24:
You didn't installed through pip?

leventelist, 2020-02-24:
No, I cloned the repo, and run the script from there.

Harvie, 2020-04-09:
Well. If you need dumb-proof way of installing bCNC. Then use pip. If you want to fiddle, then you are advanced user/developer and have to take care of this stuff on your own. However if you install bCNC using pip, it will install all dependencies and you can then easily launch bCNC from git repository.

paulvdhoeven, 2020-04-21:
First: I'm very pleased to find that bCNC works with Python3 now.
I also run in this exact issue by cloning https://github.com/vlachoudis/bCNC
I've programmed some C stuff for 25 years, but am pretty new to Python, hardly know what pip is.
The installation instructions on the main page here are for pip2, which I do not have installed. Neither do I have tkinter for Python2 installed and I do not really want to install that old software, and that was my reason for running bCNC directly with Python3.
Now I just did my first jog on my CNC for about half a year, and am curious to the improvements. bCNC seems to be pretty active.

Installing with pip3 (from my locally cloned directory) also fails with:

paul@dualcore:~/projects/python3/bCNC$ pip3 install .
Processing /home/paul/projects/python3/bCNC
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-xx0s4re0-build/

(Just to be clear: I can start bCNC & jog the axes, so communication also works with P3. It's just the install script with pip3 that encounters some error).

Harvie, 2020-04-22:
pip2 installs the package to python2, pip3 installs to python3. Other than that there's not much difference. pip without number uses whatever python your system uses by default.

So by `pip install bCNC` it will automaticaly download and install bCNC. (no need to clone git if you don't plan to modify bCNC code). Pip automaticaly downloads installed software from pypi website https://pypi.org/project/bCNC/

Harvie, 2020-04-22:
Also you can try to do `pip install --upgrade pip` and `pip install --upgrade setuptools` if your system uses old version of these.

usermicrodevices, 2020-09-23:
setup with
`pip install --upgrade bCNC`
same error
```
.local/lib/python3.7/site-packages/bCNC/FilePage.py", line 347, in comportRefresh
    for hw in i[2].split(' '):
AttributeError: 'NoneType' object has no attribute 'split'
```

usermicrodevices, 2020-09-23:
fixed with
`sudo cp -r /usr/local/lib/python3.7/dist-packages/serial .local/lib/python3.7/site-packages/`