vlachoudis/bCNC #1364: Moving over MPos 0 after Tool change

ManuelW77, 2020-02-25:
Hey guys,
i have set up bCNC (on my grbl machine) to catch the M6 commands and handle it with a
WCS tool change. The macro works fine and it probed the correct height for the tool to the 0
of the workpice.

But, after the tool change, when the spindle is moving from the tool change position to the work position and the new tool is longer then the last one, the Z axes is moving above the MPos 0 and over the homing switch. So the stepper slips and the Z position is no more correct.

It means in my opinion, if the new WPos 0 is higher then the last one, Z is moving to high, over MPos 0, and crashes.

kfmut, 2020-03-07:
@ManuelW77 

There no support for tool changing in GRBL so macro in bCNC is just workaround. You could lower your tool change height to accomodate difference between longest and shortest tools(if you are using G28 safe position then it's also should be lowered if my memory serves me well). Also soft limits in GRBL is massive help in preventing crashes you should definetly check them.

DeanRM, 2020-03-20:
Check bottom of #1371 for a possible answer to your problem.

kfmut, 2020-03-20:
@DeanRM 

You are trying to solve wrong problem. Everyone is taking about returning tip of new tool to last known WPos ;-)

DeanRM, 2020-03-21:
No - I'm not trying to solve the wrong problem.  He is talking about the z height being to high hitting his limit switch throwing off accuracy.  The settings he needs to adjust is in the User File under [Probe]. <br>
* toolprobe X, Y are the centered coordinates of the fixed/installed touch plate.
* toolprobe Z is the starting position of probing.
( Formula: [Negative #] Travel distance of Z0 (Home)[no bit installed] to the touchplate - minus the length of your longest bit)
* tooldistance is the length of movement before you get the movement error.
(Formula: The length of your longest bit plus a couple of mm)<br>
After you make a change - You must restart bCNC. The .bCNC file is only read during startup.

toolprobe Z + tooldistance should not exceed GRBLs max travel distance for Z. It also should be a little longer than the travel distance of Z0 (Home) to the touchplate.

kfmut, 2020-03-21:
@DeanRM 

So why it's too high  AFTER tool change cycle if all but last two moves in tool change macro are done in G53 space?

DeanRM, 2020-03-21:
Good question - what is his current setup and what are his current settings?

kfmut, 2020-03-21:
@DeanRM 

If you are really interested in this situation you could check old and bit long dicussion  https://github.com/vlachoudis/bCNC/issues/1029

DeanRM, 2020-03-21:
Interesting read - No where does anybody mention the proper setup for tool changes.  There are two soft limits involved - GRBL's Z and tooldistance.  Both will throw a soft limit error.  
I wrote a paper on this - https://github.com/vlachoudis/bCNC/wiki/Installing-a-fixed-Touchplate-for-tool-changes