vlachoudis/bCNC #1366: floating point exception

jrlandau, 2020-03-03:
Building a clean system on a VirtualBox VM using Debian and Python 2.7,  bCNC terminates immediately with a floating point exception. I do not see the usual python trace.
Here is what I did:
Installed Debian 10 on VirtualBox 
Enter "python", result is 2,7,16
Enter "python3", result is 3.7.3
The app sudo was installed already
added myself to sudoers 
installed pip using sudo apt install python-pip
added tkinter with sudo apt-get install python-tk
installed bCNC with pip install bCNC
added to path: /home/joe/.local/bin
Start bCNC with python -m bCNC;; got the following 2 lines:
self.canvas.winfo id(): 23068762
floating point exception

jrlandau, 2020-03-04:
Repeated the experiment with python 3 instead of python 2.  Got the same result.

jrlandau, 2020-03-07:
I am closing this issue as I am no longer trying the VirtualBox approach.

closed 2020-03-07 by jrlandau

Harvie, 2020-04-09:
I don't really understand how launching bCNC in virtualbox could be different from launching it directly...

jrlandau, 2020-04-10:
That makes two of us.  I posted the issue primarily to document that the
problem occurred.