vlachoudis/bCNC #1367: WHB04B-6 Pendant support

LYHTSPD, 2020-03-05:
I am trying to figure out is bCNC supports the WHB04B-6 manual pulse generators natively, or is that something I will have to add on my own?

Harvie, 2020-04-09:
Probably not. Unless it can pretend it's keyboard.

LYHTSPD, 2020-04-09:
I have found other CNC software that has set up a HAB configuration for them, but I am unsure of how that works.  I am still doing some digging, but could use some help.

Harvie, 2020-04-12:
bCNC has web pendant. Which also means you can write external script which communicates with whatever special hardware you have and later sends commands to bCNC over HTTP connection. You can even send custom g-code through it, if i remember correctly.