vlachoudis/bCNC #1369: Good day! I am currently testing the implementation of GRBL for the ESP32 chip. I connect to the chip via socket: // grblesp: 23 from the bCNC program. When starting the file for milling after 40 seconds - 1 minute 30 seconds, an error occurs: connection failed ([Errno 104] Connection reset by peer). I will ask for help in solving this problem. Although, when performing a surface level sample, such a problem does not arise.

isoskyflyer, 2020-03-16:


tschundler, 2020-12-30:
Does the same gcode work fine over serial? It's possible that something you are sending is triggering a reset on the controller due to an issue on the ESP side.
I found it also to be flaky for wifi sending though I don't think the issue is on the bCNC side. https://github.com/bdring/Grbl_Esp32/pull/713 can help wifi behavior.

Harvie, 2020-12-30:
I can reccomend https://github.com/terjeio/grblHAL/ I've recently tested it and it works with bCNC.
it works on ESP32 as well as on lots of other CPUs. However the WebUI is not ready yet.

Harvie, 2020-12-30:
> However the WebUI is not ready yet.

Actualy i made mistake when building it. the webui might work... i will test that later.

isoskyflyer, 2021-01-08:
Thanks a lot!

closed 2021-01-09 by Harvie