vlachoudis/bCNC #1370: WorkSpaces wiki page problems

jrlandau, 2020-03-18:
There are several problems on this page.  The first is that the formatting has gone awry, possibly due to a markdown failure.  Fixing this would be the first step, and I would be happy to do that on a branch which you could pull. However, I don't know where the file for this wiki page is; perhaps I need permission to access it. 

 It may be that I could also help in other ways.  Please let me know.

Joe

Harvie, 2020-04-09:
As far as i know, every github user should have permissions to edit the wiki. Let me know if you can't...

jrlandau, 2020-04-10:
My mistake.  I was looking for something analogous to the way one modifies
code--I didn't realize it was so simple.  I have fixed a few formatting
errors and removed a broken link.

But I have not changed the instructions for setting a workspace, which
don't make sense to me.  The 3 boxes at the bottom of the page  (right
side) don't accept entry, and the instruction to set the workspace
coordinates by using the mouse--well, how is that supposed to work?

Thanks,  Joe