vlachoudis/bCNC #1371: Edit bCNC Toolchange Script

ManuelW77, 2020-03-19:
Hey there,

is it possible to change manually the Toolchange Script?
I searched for the code but can't find it.

Regards
Manuel

MARIOBASZ, 2020-03-20:
Yes, in the cnc.py file
What are you doing?
Greetings,Mario

ManuelW77, 2020-03-20:
My problem are, if i change between tools with different lengths, the height of the work coordinate of z increases. So i'm running sometimes in a error halt, because the machine try's to move over my soft max distance in z. 

My tool change height is set to the highest point of my machine an i want to reset the work height on every tool change to this.

DeanRM, 2020-03-20:
No need to change the script.  The setting to change is in the [probe] section of the configuration file.  Your config file is located at $HOME/.bCNC.  This overrides the bCNC.ini file.  See pic.
[image]

All settings are based on machine coordinates (After homing) and mm.
toolchange X, Y, Z are the coordinates of where you want to change the tool. (Usually front and center)
toolprobe X, Y are the coordinates of the fixed/installed touch plate.
toolprobe Z is the starting position of probing.
tooldistance is the length of movement before you get the movement error.  (Formula:  Total distance of Z motion - Toolprobe Z [positive number] -  the length of your longest bit plus a couple of mm) [toolprobe Z + tooldistance should not exceed GRBLs max travel distance for Z]