vlachoudis/bCNC #1372: Motion Enable/Disable button

Git-Bruno, 2020-03-22:
Hello,
Would it be possible to add a button to enable and disable gantry motion in ordre to prevent unexpected moves. 
I got unexpected moves  when I edited gcode and I hit the up/down key as I wanted to move up/down one line in the gcode. Unfortunatly I forgot my last click was done in the graphic view area and looking at the screen I cannot tell if the up/down arrow key will move the selected line in the gcode or move the gantry.  Same occured when I change parameters in items in the 'CAM' panel.

Of course I might stop the communication to the grbl controler prior to doing any changes or use $C to get this done, but having a button would be more convenient and probably add to safety.
Many thanks for this wonderful job.
Bruno

DeanRM, 2020-03-23:
GRBL supports - Pause, Resume and Abort.  Usually pins on your control board.

Git-Bruno, 2020-03-28:
Hi, I tried using Pause/Resume Abort controls. 
I found that if I use Pause in the control panel then there is no move if I hit the up/down keys while the graphic view is 'active' in the Editor panel. BUT, all actions on the keys seem to be buffered and corresponding moves are executed when I do resume in the Control panel...not good at all!
In fact the safest way for me is to put the controler in deep sleep prior to go work in the Editor panel. 
Doing so I did not see unexpected moves after I exited deep sleep mode with the Stop control.
Thanks
Bruno