vlachoudis/bCNC #1374: Esp8266 Wifi<-> SerialBridge. No Connection with BCNC.

achest, 2020-03-29:
Hello!

I Try to use Arduino CNC wireless. I have made the schematic like :https://github.com/arkypita/ESP8266-SerialTelnet  or  similar Socket firmware.

The Hardware works. If I test it with Telnet or with websocket client, It works. Grbl answer to test messages.
But the Library in Bcnc could not establish connection .

Question: Which  firmware for ESP8266 you are recommend to use? 

I will submit Error text later.

achest, 2020-03-29:
"Remote does not seem to support RFC2217 or BINARY mode {!r}".format(mandadory_options))
serial.serialutil.SerialException: Remote does not seem to support RFC2217 or BINARY mode [we-BINARY:False(INACTIVE), we-RFC2217:False(REQUESTED)]

dylanpil, 2021-12-03:
I used ESP-link on an arduino Mega with esp on it for wifi. it worked fairly well. You can configure your ip address and other things in its gui, only after you connect to it over wifi. There is a consul where you can see when you connect to grbl, and you can send commands from here too.
and then for GRBL port I typed the ip address eg. socket://192.168.0.1:23
https://github.com/jeelabs/esp-link
Good Luck!