vlachoudis/bCNC #1375: Controlling GRBL from python scripting

flemish4, 2020-04-01:
I'm trying to control GRBL from bCNC python scripting. I'm trying to interleave actions controlled with the gpio on the rpi running bCNC with GRBL movements.

I am using the python scripting successfully however I am stuck on controlling GRBL.
I thought I should be able to execute GCODE from inside the scripting but I can't find a way to do this... 

From the commandline app.executeGcode works, but when the same code is executed from a gCode file, app is not available. The wait command also doesn't work.

Am I going about this the wrong way? 

Further info:
I have a tool changer hooked up to the rpi which is working but I need to be able to do one step with the changer, then move, then another step. Ultimately I would like to be able to generate my gcode files to include the tool changing script to have fully automated runs. I may also integrate bit inspection...

Any help appreciated.
Thanks

DeanRM, 2020-04-02:
I really can't help you much but I do have a suggestion.
What you are trying to do sounds like what was written to do a calibrate for a tool.
In ProbePage.cy - When the calibrate button is clicked, it runs this code.  Maybe it will give you a clue on how to setup your code and run it to GRBL.

	def calibrate(self, event=None):
		self.set()
		if self.check4Errors(): return
		lines = []
		lines.append("g53 g0 z[toolchangez]")
		lines.append("g53 g0 x[toolchangex] y[toolchangey]")
		lines.append("g53 g0 x[toolprobex] y[toolprobey]")
		lines.append("g53 g0 z[toolprobez]")
		if CNC.vars["fastprbfeed"]:
			prb_reverse = {"2": "4", "3": "5", "4": "2", "5": "3"}
			CNC.vars["prbcmdreverse"] = (CNC.vars["prbcmd"][:-1] +
						prb_reverse[CNC.vars["prbcmd"][-1]])
			currentFeedrate = CNC.vars["fastprbfeed"]
			while currentFeedrate > CNC.vars["prbfeed"]:
				lines.append("%wait")
				lines.append("g91 [prbcmd] %s z[toolprobez-mz-tooldistance]" \
						% CNC.fmt('f',currentFeedrate))
				lines.append("%wait")
				lines.append("[prbcmdreverse] %s z[toolprobez-mz]" \
						% CNC.fmt('f',currentFeedrate))
				currentFeedrate /= 10
		lines.append("%wait")
		lines.append("g91 [prbcmd] f[prbfeed] z[toolprobez-mz-tooldistance]")
		lines.append("g4 p1")	# wait a sec
		lines.append("%wait")
		lines.append("%global toolheight; toolheight=wz")
		lines.append("%global toolmz; toolmz=prbz")
		lines.append("%update toolheight")
		lines.append("g53 g0 z[toolchangez]")
		lines.append("g53 g0 x[toolchangex] y[toolchangey]")
		lines.append("g90")
		self.app.run(lines=lines)

MARIOBASZ, 2020-04-02:
You can send an example of the order you send him; Is it for the case of Arduino (without experience in RPI)? Maybe it will help me to continue learning? Another curiosity: what kind of changer is it?
I am on a similar path, because of quarantine I will delay the completion of the project, but the machine has already made the first movements to release the tool, take the tool, calibrate, continue. I advance very slowly.

flemish4, 2020-04-02:
Thanks for the suggestion @DeanRM, if I am able to access 'app' from a plugin I may be able to use that but I am unusure what variables are available to plugins. 

Below are some further details on my project if anyone is interested.

So far I only have the spindle lock portion made. It uses a screw, a linear potentiometer and a spring to move a spanner back and forth. The motor control and potentiometer feedback are handled by the raspberry pi. The pi is also connected to an arduino running grbl. The idea is to spring the spanner against the spindle (pi python script), then very breifly enable the spindle (pi - grbl) to get the spanner to engage. The spanner can also be used as a contact down to the bit for probing pcbs.

I have a script where this all works but I can only get it to run from the command line because I can pass 'app' into the function, whereas from gCode I cannot.

The next step will be a bank of sockets each containing a bit. The spanner will lock the spindle, then it will drive over to an empty socket position and the bit will be released etc etc. The sockets will need to be spring loaded to find the correct orientation.

[image]

flemish4, 2020-04-04:
The above fix allows me to use app.executeGcode, app.running, app.sio_wait etc. from python in a gcode file.

MARIOBASZ, 2020-04-04:
@ flemish4, unfortunately I lack knowledge to fully understand the usefulness that can give app.executeGcode, app.running, app.sio_wait, etc.
Are you doing an XATC?

flemish4, 2020-04-04:
My plan is similar to XATC but much more compact because it is a pcb mill.

Having access to the app object means I can fully control grbl from python
scripting. This allows me to time operations triggered by python - pi gpio
- tool changer with grbl movements.

So for the spindle lock shown in the picture above, I spring the spanner
against the spindle using gpio, then I turn on the spindle briefly so that
the spanner snaps into place, then the spindle can fully engage.

Hope that makes sense.

flemish4, 2020-04-04:
Okay so using app.executeGcode doesn't really work when running from gcode because it is trying to execute commands on the queue that the script is running on... So looks like this is a job for a plugin.

PR #1378 (2020-04-09): Scripting now always has access to app
Fixes #1375
This change:
Improved scriptability #1256
https://github.com/vlachoudis/bCNC/commit/4456ad2abe63121899e4833f0b3b04323d6e278c

Added the ability for scripting to access 'app', however it only added this for the commandline.
The pull request adds the functionality in two other places.

- sender.py:200           - commandline
- CNCCanvas.py:1817 - opening gcode and rendering
- sener.py:710             - running
closed 2020-04-09 by Harvie