vlachoudis/bCNC #1376: bCNC-GRBL stop communicating anytime

fbonin5491, 2020-04-02:
Hi , i have made a lot of testings  , with / without motors i made my own shielded usb cable to be sure its a good quality one , and its still the same , stopping anytime , 10% , the most 48%.I did try also UGS , same result , its very frustrating.i never had this problem before , as i was only doing some PCB engraving or cutting.But image engraving...no success.
Any fresh idea?
Thank you

MARIOBASZ, 2020-04-03:
Possible electrical noise.
Arduino as close to the pc, away from the cnc. Disable hard limits.
Unused pins in the air instead of gnd.
If you solve the problem make your comment, it will help us all.

fbonin5491, 2020-04-04:
Hi , my cable is less then 1 m and properly shielded , and its not a problem of motors cable , because , without motor connected , its the same.But , **if i chose 0.25mm spot lazer size instead of 0.1mm , the work complete** ,(i made my cnc file with Gimp G-code plugin).I tested also with LazerGRBL on Win7 on a virtual machine , and it works fine in the **condition** that i choose **Synchro Mode** instead of **Buffer** , and  **Threading Mode** to **fast** instead of **ultrafast**.These 2 points tells me that the problem is the capacity of streaming data between Arduino and the PC via USB.

DeanRM, 2020-04-05:
What OS are you using?  Windows has a history of dropping the USB signal.

fbonin5491, 2020-04-05:
Linux Lite , based on Ubuntu 18.04

nealre, 2020-04-09:
I have used and enjoyed bcnc for about 1 1/2yrs. Then I put it aside for a while. During that time something has happened? I tried to upgrade to the latest and now the USB port won't connect and I have tried all I know to do. Went on and seen what others have done but my windows is not professional and for what ever reason never got the usb to load...
The bcnc program loads fine but complains about the usb. Sure could use some help.
Rick

ricardo2kK, 2020-04-10:
Yo solucione usando el programa Putty 0.73  y el pyserial este ultimo instalandolo dentro del programa   python2.7.13 (utilizo win7 stater)  este video me ayudo bastante  copio enlace 
https://www.youtube.com/watch?v=WUk7II9IJf8  Espero que te sirva  ,abrazos Ricardo

vadimyar, 2024-06-26:
[SOLUTION (possibly...)](https://github.com/vlachoudis/bCNC/issues/1808#issuecomment-2100233973)