vlachoudis/bCNC #1377: Plugin database is not expanded in some plugins

MARIOBASZ, 2020-04-03:
In some plugins the variants do not appear, for example in Driller (in Drill yes), and I do not remember or find that it is missing to add or modify somewhere to activate it.
Without being essential, I find it useful.
If anyone can help and explain to me, it is appreciated.
Hugs.

DodoLaSaumure, 2020-05-15:
Hello dear  @MARIOBASZ 

I am not sure to understand

In driller tool I have the following screenshot

[image]

It corresponds to the code, I do not see any "drill Yes" option 

		self.variables = [
			("name",          "db",    "",    _("Name")),
			("HolesDistance", "mm",    10.0,  _("Distance between holes")),
			("TargetDepth",   "mm",    0.0,   _("Target Depth")),
			("Peck",          "mm",    0.0,   _("Peck, 0 means None")),
			("Dwell",         "float", 0.0,   _("Dwell time, 0 means None")),
			("useAnchor",     "bool",  False, _("Use anchor")),
			("File",          "file",  "",    _("Excellon-File")),
			("useCustom",     "bool",  False, _("M3 for laser (settings below)")),
			("rFeed",         "int",   "",    _("Feed rapid G0"),"Defaults from config, if blank"),       # default = min(CNC.feedmax_x, CNC.feedmax_y)
			("spinMin",       "int",   "",    _("Laser power minimum"),"Defaults from config, if blank"), # default = Utils.config.get("CNC","spindlemin")
			("spinMax",       "int",   "",    _("Laser power maximum"),"Defaults from config, if blank"), # default = Utils.config.get("CNC","spindlemax")
		]

=> I am not sure to understand your issue. Do you want to add a "drill yes" variable ?
Have a good day
Pierre

MARIOBASZ, 2020-05-15:
Hi Pierre, I actually wrote something simple in a confusing way:
The name of the plugin variant is not saved; when you make another variant with other parameters it gets overwritten (let's suppose you want to repeatedly use the anchor sometimes, another option at others).

DodoLaSaumure, 2020-05-15:
When I click on the add button I get a new driller
Is this ok for you Mario ?

[image]

MARIOBASZ, 2020-05-15:
It is exactly that.
Can you tell me what you have modified? I do not remember how I did it in Helical and others, to use it in future plugins.
By the way, maybe I need help with complement to use two milling cutters (roughing (avoid overlapping of adjacent figures) and termination (clean sectors not treated by milling cutter of greater diameter)), will you? I've made progress but I'm still missing (probably I should learn and use shapely, it's very slow) # 988

[image]

DodoLaSaumure, 2020-05-15:
Dear Mario
The variable
`("name",          "db",    "",    _("Name")),`
provides a name to the plugin and adds it into the db, in the line
`		self.variables = [.....]`

I found  exceptions that were raised in tkExtra.py, depending of the version of Python used, please look at modifications in  28b02d6 => I am going to create a pull request on this commit.

I am going to have a look at #988 and see if I can help.

Have a good day
Pierre

Harvie, 2020-12-28:
Reportedly fixed in this PR. Please test and close this issue if everything is OK:
https://github.com/vlachoudis/bCNC/pull/1405#issuecomment-751580531

closed 2020-12-28 by MARIOBASZ