vlachoudis/bCNC #1379: Resize the gcode en bCNC

gnurc, 2020-04-05:
Hello,

I'm using Inkscape and a plugin for raster but this plugin resize the object. It's possible to resize the gcode in bCNC?

It's possible to raster a .svg in bCNC in order to not use the Inkscape plugin?

Best regards

MARIOBASZ, 2020-04-05:
scale plugin can resize

gnurc, 2020-04-05:
Hi,

I want resize this circuit,
[image]

But the result is this
[image]

MARIOBASZ, 2020-04-05:
I think the problem is all in one height Z:
Test first by making a cut (example cut -0.1 0.1) and then scale

MARIOBASZ, 2020-04-05:
[image]
Scaling 0.92 - 0.92

gnurc, 2020-04-05:
Thank you, that works.

But, How can I get the surface back to 0?  When I try, the object disappears.

MARIOBASZ, 2020-04-05:
If I remember correctly, it is related to the value of Z safe: if it is above Z safe, G0, below G1, you could modify the Z safe height, although I do not recommend it.
I don't know the practical value of taking Z0: a solution that might be useful to you is cut to Z-0.001.
I'm glad it's useful.
Greetings.

MARIOBASZ, 2020-04-05:
Experience Flat Path plugin

gnurc, 2020-04-06:
Yes, it's related with the Z, but by the moment I have done the scale with http://www.scorchworks.com/Gcoderipper/gcoderipper.html#download  and works without problems in bCNC.

Thank you

gnurc, 2020-04-06:
I have solved added Z0.0 in the gcode.

Thank you.

closed 2020-04-16 by gnurc