vlachoudis/bCNC #1380: Crash on Mac 10.14.6 / bCNC bCNC-0.9.14.304

MikeMitterer, 2020-04-09:
I tried it with python 2.7 and 3.8 (via Anaconda Environment) 
bCNC crashes the whole machine. It logs me out!!! - I've never seen this before on my Mac and I use Mac since x-years.

Here is the crash report if this helps:
```bash
Process:               AdobeGCClient [13649]
Path:                  /Library/Application Support/Adobe/*/AdobeGCClient.app/Contents/MacOS/AdobeGCClient
Identifier:            AdobeGCClient
Version:               6.4.0.55 (6.4.0.55)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
Responsible:           AdobeGCClient [13649]
User ID:               501

Date/Time:             2020-04-09 15:39:52.921 +0200
OS Version:            Mac OS X 10.14.6 (18G1012)
Report Version:        12
Anonymous UUID:        9F8EFB51-517C-E475-C614-3755DF27ABD4

Time Awake Since Boot: 29000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [13649]

VM Regions Near 0x8:
--> 
    __TEXT                 00000001077ee000-0000000107bed000 [ 4092K] r-x/rwx SM=COW  /Library/Application Support/Adobe/*/AdobeGCClient.app/Contents/MacOS/AdobeGCClient

Application Specific Information:
LaunchServices: registering extension point com.apple.textinputmethod-services at URL file:///System/Library/Frameworks/InputMethodKit.framework/

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff47fd5e9f CFMessagePortInvalidate + 20
1   com.adobe.gcclient            	0x0000000107a2e2fb 0x1077ee000 + 2360059
2   com.adobe.gcclient            	0x0000000107a2ecfc 0x1077ee000 + 2362620
3   com.adobe.gcclient            	0x0000000107a3650f 0x1077ee000 + 2393359
4   com.adobe.gcclient            	0x0000000107a4300b 0x1077ee000 + 2445323
5   com.adobe.gcclient            	0x0000000107a377e3 IAL_GetClientConfiguration + 69
6   com.adobe.gcclient            	0x0000000107820592 0x1077ee000 + 206226
7   com.adobe.gcclient            	0x00000001078c9003 0x1077ee000 + 897027
8   com.adobe.gcclient            	0x000000010788f166 0x1077ee000 + 659814
9   com.adobe.gcclient            	0x000000010784cb24 0x1077ee000 + 387876
10  com.adobe.gcclient            	0x000000010784c7db 0x1077ee000 + 387035
11  com.adobe.gcclient            	0x00000001078c1c88 0x1077ee000 + 867464
12  com.adobe.gcclient            	0x000000010783a5bb 0x1077ee000 + 312763
13  com.adobe.gcclient            	0x0000000107839f0c 0x1077ee000 + 311052
14  com.adobe.gcclient            	0x000000010783972a 0x1077ee000 + 309034
15  com.adobe.gcclient            	0x00000001077f4834 start + 52

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff741213f0 start_wqthread + 0

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff74064bfe __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff741213db _pthread_workqueue_addthreads + 40
2   libdispatch.dylib             	0x00007fff73eeb0d8 _dispatch_root_queue_poke_slow + 172
3   libdispatch.dylib             	0x00007fff73ef319d _dispatch_source_invoke + 2659
4   libdispatch.dylib             	0x00007fff73ee7792 _dispatch_lane_serial_drain + 268
5   libdispatch.dylib             	0x00007fff73eeff73 _dispatch_kevent_worker_thread + 990
6   libsystem_pthread.dylib       	0x00007fff7412161b _pthread_wqthread + 431
7   libsystem_pthread.dylib       	0x00007fff741213fd start_wqthread + 13
[... 388 lines cut ...]
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Activity Tracing                   256K        1 
CoreServices                       256K        1 
Dispatch continuations            16.0M        1 
Kernel Alloc Once                    8K        1 
MALLOC                            72.7M       20 
MALLOC guard page                   16K        3 
PROTECTED_MEMORY                     4K        1 
STACK GUARD                       56.0M        6 
Stack                             10.5M        6 
VM_ALLOCATE                          4K        1 
__DATA                            40.3M      319 
__FONT_DATA                          4K        1 
__LINKEDIT                       227.6M       10 
__TEXT                           342.7M      314 
__UNICODE                          564K        1 
mapped file                       26.0M        3 
shared memory                     3576K        7 
===========                     =======  ======= 
TOTAL                            796.3M      696 

```

Harvie, 2020-04-09:
Well that's weird. Whatever you do in python code should not crash whole machine. While i am not telling you that bCNC code is 100% perfect, i feel like there might be some other issue with your system as well...

Harvie, 2020-04-09:
TBH IDK what AdobeGCClient process is. It does not seem related to bCNC at all...

MikeMitterer, 2020-05-17:
I nailed it down to tcl/tk - neither the one installed via brew nor the other one, installed via pip, works. If I run a simple tcl/tk test it either logs me out or crashes my Mac