vlachoudis/bCNC #1389: Unexpected move after stop

calabr, 2020-04-21:
bCNC 0.9.14
py3.8.1
 
After probe cycle interrupt by pressing "Stop" button, bCNC send 
G53 G0 X0 Y0 Z0
to the GRBL
This behaviour only found in Probe sequence. Tool calibrate and tool change
didn't send this command.

 Terminal listing:

ok
Grbl 1.1f ['$' for help]
$#
[G54:-223.130,-102.484,-91.392]
[G55:-287.000,-181.000,-77.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:-287.000,-181.000,-1.000]
[G30:-287.000,-1.000,-1.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
b'$G\n'
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$X
ok
G0 G54 G17 G21 G90 G94
ok
G43.1Z0.000
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
G53 G0 X0 Y0 Z0
ok

ok
$#
[G54:-223.130,-102.484,-91.392]
[G55:-287.000,-181.000,-77.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:-287.000,-181.000,-1.000]
[G30:-287.000,-1.000,-1.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

GitHubCaps, 2020-04-25:
Just noting that this also relates to #1216
Also, b'$G\n' should not be there (needs more code cleanup to be compatible with python 3)

calabr, 2020-04-25:
I found the move to X0Y0Z0 cused by tetting "Automatic GOTO after probing" check-box.
If probbing cycle interrrupted, variables prbx, prby, prbz  contain's 0 0 0 and after probbing interrupted, flag 
self.probeautogotonext = True
 and than goto2Probe method called
 I can't figure out why to use this combo-box. In my first try I think it should return to the position before begining of probe cycle.
  With current bihaviour it's trying to move to the same possition, whrer we already stopped, so do nothing after success probe.
   To owecome this behaviour, I think it's need to clear this flag during stop sequence.
  I can't find right way to do it. Decided just disable self.goto2Probe() execution.

micksulley, 2020-04-27:
I just tested on my system, start a probe and while it is inching down switch to the Control tab and press Stop.  It stops, pause for about 1/2 second and then drives off in positive X and Y directions.  This is the same result as I reported in Unexpected move following alrm reset #1216.  
If you need me to carry out any other tests just let me know, I'll be happy to help if I can.
Mick

kfmut, 2020-05-26:
@calabr 

> I can't figure out why to use this combo-box. In my first try I think it should return to the position before begining of probe cycle.
> With current bihaviour it's trying to move to the same possition, whrer we already stopped, so do nothing after success probe.

Actually probed position and stopped position are not same, depending on feedrate for G38 move they could be different for about 0.1-0.2mm, that check box really helps when acccurate zeroing is must, like PCB milling.