vlachoudis/bCNC #1390: ProbePage.py: Fix a few issues (hopefully)

GitHubCaps, 2020-04-23:
There is a typo that will always return 0.
line 1802 in ProbePage.py:
`		Utils.setFloat("Probe", "toolmz",      CNC.vars.get("toolmz",0.))`
should be:
`		Utils.setFloat("Probe", "toolmz",      CNC.vars.get("toolmz"))`
which is causing it to act more like a `setter` than a `getter`.
This will cause a lot of issues and is most likely contributing to many issues such as:  #1371, #1364, #1347, #1281, #1140, #1035 to name a few...
This may not solve all those problems, but should be corrected nonetheless.
@Harvie  I have been having a lot of issues with github on my pc lately (and my internet connection) and have refrained from doing any commits. Please, if you or someone else doesn't mind merging the change, thanks!

Harvie, 2020-04-23:
It seems to me that 0. is only default value which gets returned when there is no actual value stored in CNC.vars, but i wasn't able to locate the code of the getter yet... Do you know where it is? Maybe there is bug which causes it to always return default value??

GitHubCaps, 2020-04-23:
The better is not what is important here. What happens is that no matter what you set the value to will not be taken into account, instead it will always return the wrong value and that you simply don't want.

GitHubCaps, 2020-04-23:
That was supposed to be getter, not better

GitHubCaps, 2020-04-23:
If you want to return a default value, it would be better to use:
`Utils.setFloat("Probe", "toolmz",      CNC.vars.get("toolmz",""))`

Harvie, 2020-04-23:
Yes. I need to see where  CNC.vars.get() method is defined to get better understanding of this... But i can't find it right now.

GitHubCaps, 2020-04-23:
The getter is actually from Utils.py:
`def setStr(section, name, value):`

`setFloat = setStr`

GitHubCaps, 2020-04-23:
It is a value that does not exist until the user creates it, then you grab it from...

GitHubCaps, 2020-04-23:
Has nothing to do with CNC.vars per se

Harvie, 2020-04-23:
But i am looking for get() not for setStr()... is that automaticaly created by python?

GitHubCaps, 2020-04-23:
Later on it gets set, that part works correctly after fixing the typo in ProbePage

GitHubCaps, 2020-04-23:
The typo in ProbePage causes it to always fail no matter what

Harvie, 2020-04-23:
Yes. But the typo is
`CNC.vars.get("toolmz",0.)` vs. `CNC.vars.get("toolmz")`

I want to know, where that .get() method is defined.

GitHubCaps, 2020-04-23:
Sorry, should have explained that better... I am only referring to how it behaves, not that it uses the __getter__ or __setter__ methods

Harvie, 2020-04-23:
And i believe you that there is a problem which is fixed by this. But. I first need to see the code, to undersand what caused the bug, so it does not happen again in the future.

GitHubCaps, 2020-04-23:
It should return correct value after creation, updating etc. regardless. Keep in mind that it didn't exist to begin with, however upon creation, it needs to return the correct value, not 0.

Harvie, 2020-04-23:
Yes. That's why i think, we should directly fix the code of getter method, rather than just its call.

GitHubCaps, 2020-04-23:
Not a getter method. It is a variable that is directly used by CNC.toolPolicy==3 for TLO and relies solely on Utils and ProbePage, no other instances really exist.

GitHubCaps, 2020-04-23:
Oddly enough, it doesn't even exist as a variable in CNC.vars, maybe it should be added there. There is also two references to TLO (as a variable) in CNC.vars when there should be only one.

GitHubCaps, 2020-04-23:
In CNC.py line 710 and 752 both have TLO, line 719 is lowercase and is used differently, which is fine. A variable could be added as a default for `toolmz`, but aren't any for `toolmx` or `toolmy` either. Like I was saying, these are created by the user and are set/get by Utils/CNC/ProbePage only.

GitHubCaps, 2020-04-23:
If I had to guess the get method you where asking about, I would say it is the standard python get method for grabbing, not something that bCNC is using. Utils is setting the value after grabbing the default (which doesn't exist) and setting it accordingly (after the user inputs it, now it exists)

GitHubCaps, 2020-04-23:
I know I probably confused you in more ways than one, but the gist of it is that it basically wasn't an option or variable to begin with, but once the user does create it, it will never return the correct value because of that simple typo. I don't even use tool offsets myself on my machine, but while I have been sorting through the code and trying to resolve some other issues I discovered that in the toolchange and it literally will screw up everything by returning 0 instead of the value that the user sets.

GitHubCaps, 2020-04-23:
Oh man,

> It seems to me that 0. is only default value which gets returned when there is no actual value stored in CNC.vars, but i wasn't able to locate the code of the getter yet... Do you know where it is? Maybe there is bug which causes it to always return default value??

I see where you are going with that! There is no other .get() methods besides in ProbePage.py, strange. May be that things got changed way back and never caught.
Even `updateTlo` and `updateProbe` uses .get(), but how or why hasn't this been caught before?

As far as:
> Maybe there is bug which causes it to always return default value??

well, in my simple test, fixing the typo at least, helped to retrieve the correct value back, just scratching my head like a dumb monkey at the moment...

Scratch that, it is using `__getitem__` in CNC `return CNC.vars[name]`, the rest seems to be handled by `config.get()` and if it doesn't exist, `Utils.setStr` is using `config.set(section, name, str(value))` to set it. Hope that helps!

JipeB, 2021-10-04:
Hello,
Sorry to dig up this already old topic, but I'm using The Tool Change (TLO).

"line 1802 in ProbePage.py:
Utils.setFloat ("Probe", "toolmz", CNC.vars.get ("toolmz", 0.)) "

This code works perfectly, it is used when changing the tool. "toolmz" is initialized when starting bCNC from the .bCNC file and stored in the variable CNC.var ["toolmz"]; it is forced to 0 if the variable in the .bCNC file is blank.
"toolmz" makes it possible to find, at each start of bCNC, an absolute Z coordinate which serves as point 0 to measure the length of the tool that is going to be used (TLO).

As I have read on several occasions, the TLO method is not recommended due to exceeding the software limits, and yet, if the G43.1 function is understood and correctly applied there is no risk of going out of limits. This is also the only method used on professional machines.

JP