vlachoudis/bCNC #1392: New .exe binary

PT400C, 2020-04-25:
Searching for the latest windows binaries, I only find precompiled versions from 2018. Could you release a compiled .exe of the latest [version](https://github.com/vlachoudis/bCNC/releases/tag/0.9.14.304) (19 days ago)?

Thanks in advance

Harvie, 2020-04-25:
Why don't you install bCNC using pip?
BTW once installed using pip, there is batch file to produce .exe if i remember correctly. But currently i don't feel like it's really needed. I might create automated builds later, but i am waiting for travis to support python...

Harvie, 2020-04-25:
https://github.com/vlachoudis/bCNC/wiki/Installation

PT400C, 2020-04-26:
@Harvie Thanks so much for your answer. I actually did that (compiling the exe myself), but always thought I had an old version. It always stated that the last update was in January 2019 - indicating that I had compiled an old version of the sourcecode. But actually this seems to be a bug - I eventually was able to compile the latest source.

closed 2020-05-15 by PT400C