vlachoudis/bCNC #1393: Restart spindle after manual tool probe

gauna85, 2020-04-26:
For manual toolchange the following code is appended:

		lines.append("f[feed] [spindle]")# ... feed and spindle
		lines.append("g4 p5")		# wait 5s for spindle to speed up

where last spindle state is restored after toolchange, which is very dangerous, if maximum allowed rpm for the new tool is lower than the max speed of last tool.

MARIOBASZ, 2020-04-26:
I think it is fine as it is: if there was an error in choosing a too high speed, there was already for the tool before the change.
But if I'm not mistaken, if in grbl the maximum S is 12000 and you send 24000, it will turn to 12000

gauna85, 2020-04-26:
The speed before the tool change corresponds to the old tool e.g. 20000 for a 6mm diameter. After tool change this speed applied again during restore. An, of course, my limit is not set to 12000. I'm not talking to a "manual" toolchange by clicking in the toolbar. I'm talking about a "manual" toolchange within a gcode program, by transfering M6 cmd to a specific sequence by bcnc

MARIOBASZ, 2020-04-26:
I'm missing something?
Let's see if I interpret what you are saying, after the tool change, shouldn't the new speed be placed on the next line?
suppose
T1 6mm S 20000
T2 3mm S 12000
the code should be
.......
M6 T2
S 12000
I have not found conflict in its use,

gauna85, 2020-04-27:
M6 is not send to grbl. M6 is replaced by a series of cmds. The last cmd restores spindle state stored in [spindle]. This is "S 20000" At least, I think so. Then, after this cmd, "S 12000" is read in and transfered to grbl. M6 cmd is finished by restoring the spindle state just before reading "S 12000". And between restore an "S 12000" is a delay, where the spinlde could turn up. I can observe the problem in reality and I'm looking for the reason. Of course, could also be a problem in my grbl fork. But at the moment I would say, the problem is located in bCNC. I will test is next weekend.

GitHubCaps, 2020-04-28:
@gauna85  Just for clarification, you want it to abide by what is in the gcode file and not have it issue:

```python
		lines.append("f[feed] [spindle]") # ... feed and spindle
```

which to me makes sense, since you are doing this manually with a file that probably has its own max. spindle values that should be honored instead.

What `toolPolicy` do you use? 2 for WSC or 3 for TLO? Does your code also provide for feedrate and/or `G4` (pause)?

Anyways, as an example, I think something like the following would be appropriate for your use case, but possibly could use a tickbox (use feed/spindle from file) in the toolchange section to have it optional (don't know how other users expect this to work):

```python
	#----------------------------------------------------------------------
	# code to change manually tool
	#----------------------------------------------------------------------
	def toolChange(self, tool=None):
		if tool is not None:
			# Force a change
			self.tool = tool
			self._lastTool = None

		# check if it is the same tool
		if self.tool is None or self.tool == self._lastTool: return []

		# create the necessary code
		lines = []
		lines.append("$g") # remember state and populate variables, FIXME: move to ./controllers/_GenericController.py
		lines.append("m5") # stop spindle
		lines.append("%wait")
		lines.append("%_x,_y,_z = wx,wy,wz") # remember position
		lines.append("g53 g0 z[toolchangez]")
		lines.append("g53 g0 x[toolchangex] y[toolchangey]")
		lines.append("%wait")

		if CNC.comment: # XXX: Why %% are used here, but nowhere else?
			lines.append("%%msg Tool change T%02d (%s)"%(self.tool,CNC.comment))
		else:
			lines.append("%%msg Tool change T%02d"%(self.tool))
		lines.append("m0") # feed hold

		if CNC.toolPolicy < 4:
			lines.append("g53 g0 x[toolprobex] y[toolprobey]")
			lines.append("g53 g0 z[toolprobez]")

			# fixed WCS
			if CNC.vars["fastprbfeed"]:
				prb_reverse = {"2": "4", "3": "5", "4": "2", "5": "3"}
				CNC.vars["prbcmdreverse"] = (CNC.vars["prbcmd"][:-1] +
							     prb_reverse[CNC.vars["prbcmd"][-1]])
				currentFeedrate = CNC.vars["fastprbfeed"]
				while currentFeedrate > CNC.vars["prbfeed"]:
					lines.append("%wait")
					lines.append("g91 [prbcmd] %s z[toolprobez-mz-tooldistance]"%(
							CNC.fmt('f',currentFeedrate))
					lines.append("%wait")
					lines.append("[prbcmdreverse] %s z[toolprobez-mz]"%(
							CNC.fmt('f',currentFeedrate))
					currentFeedrate /= 10
			lines.append("%wait")
			lines.append("g91 [prbcmd] f[prbfeed] z[toolprobez-mz-tooldistance]")

			if CNC.toolPolicy==2:   # manual tool change (WCS)
				# Adjust the current WCS to fit to the tool
				# FIXME could be done dynamically in the code
				p = WCS.index(CNC.vars["WCS"])+1
				lines.append("g10l20p%d z[toolheight]"%(p))
				lines.append("%wait")

			elif CNC.toolPolicy==3: # manual tool change (TLO)
				# Modify the tool length, update the TLO
				lines.append("g4 p1") # wait a sec to get the probe info
				lines.append("%wait")
				lines.append("%global TLO; TLO=prbz-toolmz")
				lines.append("g43.1z[TLO]")
				lines.append("%update TLO")

			lines.append("g53 g0 z[toolchangez]")
			lines.append("g53 g0 x[toolchangex] y[toolchangey]")

		if CNC.toolWaitAfterProbe:
			lines.append("%wait")
			lines.append("%msg Restart spindle")
			lines.append("m0") # feed hold

		# restore state
		lines.append("g90")                   # restore mode
		lines.append("g0 x[_x] y[_y]")        # ... x,y position
		lines.append("g0 z[_z]")              # ... z position

		# do not append feed/spindle for these policies
		if not CNC.toolPolicy==2 or CNC.toolPolicy==3:
			lines.append("f[feed] [spindle]") # ... feed and spindle
		lines.append("g4 p5")                 # wait 5s for spindle to speed up

		# remember present tool
		self._lastTool = self.tool
		return lines

```

Again, there's probably a better alternative, but the code before appended it regardless of policy, which in my opinion should be based off the policy/users usage of the function instead.

gauna85, 2020-04-28:
Yes, I want to skip this line.  Just for my information is "[spindel]" the last value in my gcode file or even the value in GUI? But anyway, running a gcode file should never restart the spindle by the end of M6. This is normally done in the next line. But shouldn't this be defined in a kind of reference? what is M6 doing?

I'm using TLO.

GitHubCaps, 2020-04-28:
I believe it just uses the max spindle value from bCNC configuration that you update to the controller to use:
[image]

If you have some code to test with, I could verify to be sure you get the results you are looking for.

As far as it restarting, does your file already have the `G4` to wait? Also, I just edited the code above to only omit the feed and spindle for `TLO` in case it is only needed for that specific use, may need to indent the line

```python
		lines.append("g4 p5")             # wait 5s for spindle to speed up
```
if you don't need it.

GitHubCaps, 2020-04-28:
However, it may store the last spindle rate from previous tool used or possibly even an override if you used it during the run from the GUI.

gauna85, 2020-04-28:
I generate my gcode with FreeCAD post with "--wait-for-spindle 2 --tool-change". My code includes therefor for G4 cmd
```
M6 T2.0
M3 S24000.0
G4 P2
```

Yes grbl is ignoring M6. That is why bcnc insert the sequence. I though about and "offical" gcode reference, where the functional operation of M6 is described e.g. if M6 implementation is supposed to come back to the last operating position before tool change or if the post processor has to make sure to come back to the last position if needed. Just my thought. I'm not sure if there are any standards.

I attached a simple test file generated by Freecad with 2 different tools and spindle speeds. 

Thanks for your effort. 

[test.zip](https://github.com/vlachoudis/bCNC/files/4547749/test.zip)

GitHubCaps, 2020-05-07:
I apologize about the late reply...

`[spindle]` is the mode being used, not the rpm of the spindle itself. You can handle your own tool change directly in the gcode or using a macro in bCNC. 

This is an example to show what is happening when you run M6 (without actually using it) that you can view in the editor.

```gcode
(Block-name: Header)
(Block-expand: 1)
(Block-enable: 1)
(Exported by FreeCAD)
(Post Processor: grbl_post)
(Output Time:2020-04-28 20:12:41.319659)
(Begin preamble)
G17 G90
G21
(Block-name: Fixture)
(Block-expand: 1)
(Block-enable: 1)
(Begin operation: Fixture)
(Path: Fixture)
G54
(Finish operation: Fixture)
(Block-name: T2: Endfraeser 4mm)
(Block-expand: 1)
(Block-enable: 0)
(Block-X: [Begin operation: T2: Endfraeser 4mm])
(Block-X: [Path: T2: Endfraeser 4mm])
(Block-X: [T2: Endfraeser 4mm])
(Block-X: [Begin toolchange])
(Block-X: M3 S0)
(Block-X: M6 T2)
(Block-X: M3 S24000)
(Block-X: G4 P2)
(Block-X: [Finish operation: T2: Endfraeser 4mm])
(Block-name: T2: Endfraeser 4mm)
(Block-expand: 1)
(Block-enable: 1)
(Begin operation: T2: Endfraeser 4mm)
(Path: T2: Endfraeser 4mm)
(T2: Endfraeser 4mm)
(Begin toolchange)
(set initial feedrate value?)
F600
(set initial spindle RPM to zero or expected value)
M3 S0
(This is what M6 is doing!)
(Note: there is a check for tool here that I am not handling!)
(remember state and populate variables)
$G
(stop spindle)
M5
%wait
(remember position)
%_x,_y,_z = wx,wy,wz
g53 g0 z[toolchangez]
g53 g0 x[toolchangex] y[toolchangey]
%wait
%msg Tool change T2 (4mm)
(feed hold)
m0
(if toolPolicy < 4)
g53 g0 x[toolprobex] y[toolprobey]
g53 g0 z[toolprobez]
(there is a while loop here... I am just using fastprbfeed, but can use a variable set here)
%wait
g91 [prbcmd] f[fastprbfeed] z[toolprobez-mz-tooldistance]
%wait
[prbcmdreverse] f[fastprbfeed] z[toolprobez-mz]
%wait
g91 [prbcmd] f[prbfeed] z[toolprobez-mz-tooldistance]
(end of while loop)
(wait a second to get the probe info)
g4 p1
%wait
%global TLO; TLO=prbz-toolmz
g43.1z[TLO]
%update TLO
g53 g0 z[toolchangez]
g53 g0 x[toolchangex] y[toolchangey]
%wait
%msg Restart spindle
(feed hold)
m0
g90
g0 x[_x] y[_y]
g0 z[_z]
(here, spindle represents M[mode])
(f[feed] [spindle]) ; commented out
(g4 p5) ; commented out and put into the next block
(Note: _lastTool is set here; I am not handling it though!)
%msg Finish operation
(Finish operation: T2: Endfraeser 4mm)
(Block-name: MillFace)
(Block-expand: 1)
(Block-enable: 1)
(Begin operation: MillFace)
(Path: MillFace)
(MillFace)
M3 S24000 (start spindle)
g4 p5 (wait 5 seconds for spindle)
G0 Z60.000
G1 X8.000 Y2.000 Z10.000 F600.000
G1 X8.000 Y4.829 Z10.000 F1800.000
G1 X5.171 Y2.000 Z10.000 F1800.000
G1 X2.343 Y2.000 Z10.000 F1800.000
G1 X8.000 Y7.657 Z10.000 F1800.000
G1 X8.000 Y8.000 Z10.000 F1800.000
G1 X5.514 Y8.000 Z10.000 F1800.000
G1 X2.000 Y4.486 Z10.000 F1800.000
G1 X2.000 Y7.314 Z10.000 F1800.000
G1 X2.686 Y8.000 Z10.000 F1800.000
G1 X2.000 Y8.000 Z10.000 F1800.000
G1 X2.000 Y7.314 Z10.000 F1800.000
G0 Z60.000
G0 Z60.000
(Finish operation: MillFace)
(Block-name: T3: Endfraeser 6mm)
(Block-expand: 1)
(Block-enable: 1)
(Begin operation: T3: Endfraeser 6mm)
(Path: T3: Endfraeser 6mm)
(T3: Endfraeser 6mm)
(Begin toolchange)
M3 S0
M6 T3
M3 S19099
G4 P2
(Finish operation: T3: Endfraeser 6mm)
(Block-name: MillFace001)
(Block-expand: 0)
(Block-enable: 1)
(Begin operation: MillFace001)
(Path: MillFace001)
(MillFace001)
G0 Z60.000
G1 X6.757 Y7.000 Z10.000 F600.000
G1 X3.000 Y3.243 Z10.000 F1800.000
G1 X3.000 Y7.000 Z10.000 F1800.000
G1 X6.757 Y7.000 Z10.000 F1800.000
G1 X7.000 Y7.000 Z10.000 F1800.000
G1 X7.000 Y3.000 Z10.000 F1800.000
G0 Z60.000
G0 Z60.000
(Finish operation: MillFace001)
(Block-name: post_amble)
(Block-expand: 0)
(Block-enable: 1)
(Begin postamble)
M5
G17 G90
M2
```

Also, another example is the `calibrate` function from `ProbePage.py`

```python
	def calibrate(self, event=None):
		self.set()
		if self.check4Errors(): return
		lines = []
		lines.append("g53 g0 z[toolchangez]")
		lines.append("g53 g0 x[toolchangex] y[toolchangey]")
		lines.append("g53 g0 x[toolprobex] y[toolprobey]")
		lines.append("g53 g0 z[toolprobez]")
		if CNC.vars["fastprbfeed"]:
			prb_reverse = {"2": "4", "3": "5", "4": "2", "5": "3"}
			CNC.vars["prbcmdreverse"] = (CNC.vars["prbcmd"][:-1] +
						prb_reverse[CNC.vars["prbcmd"][-1]])
			currentFeedrate = CNC.vars["fastprbfeed"]
			while currentFeedrate > CNC.vars["prbfeed"]:
				lines.append("%wait")
				lines.append("g91 [prbcmd] %s z[toolprobez-mz-tooldistance]" \
						% CNC.fmt('f',currentFeedrate))
				lines.append("%wait")
				lines.append("[prbcmdreverse] %s z[toolprobez-mz]" \
						% CNC.fmt('f',currentFeedrate))
				currentFeedrate /= 10
		lines.append("%wait")
		lines.append("g91 [prbcmd] f[prbfeed] z[toolprobez-mz-tooldistance]")
		lines.append("g4 p1")	# wait a sec
		lines.append("%wait")
		lines.append("%global toolheight; toolheight=wz")
		lines.append("%global toolmz; toolmz=prbz")
		lines.append("%update toolheight")
		lines.append("g53 g0 z[toolchangez]")
		lines.append("g53 g0 x[toolchangex] y[toolchangey]")
		lines.append("g90")
		self.app.run(lines=lines)
```