vlachoudis/bCNC #1394: Unable to install on Ubuntu 20.04 using python2/pip2

avarakin, 2020-04-26:
I tried installing bCNC on Ubuntu 20.04 and installation fails due to missing pip2. After some digging, I found that Debian and Ubuntu are dropping support for Python 2.
I was able to Install using python 3, but I think it is time to fully migrate bCNC to Python 3 and update the installation instructions accordingly.

Harvie, 2020-04-27:
Python 3 is supported by bCNC. There still might be bugs, but that applies to python 2 as well. Basicaly this is only matter of documentation and IIRC it's already mentioned that you can use whatever python/pip version you prefer...

bepitama, 2020-05-10:
In Ubuntu 20.04 using python2 at startup gives me the error: Please install the python pyserial module ...
Python-serial is no longer included in Ubuntu 20.04.Trying to launch it in Python3 on Ubuntu 20.04 gives me this error:
giuseppe@giuseppe-HP-250-G5:~/opt/bCNC-master$ python3 bCNC.py
Traceback (most recent call last):
  File "bCNC.py", line 57, in <module>
    from CNC import WAIT, CNC, GCode
  File "/home/giuseppe/opt/bCNC-master/CNC.py", line 3728
    print "cmds=",cmds
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("cmds=",cmds)?
Thanks
Giuseppe

bepitama, 2020-05-11:
It works. I solved it by manually installing the python-serial_3.4-4_all.deb file.
To install the python 2 libraries, on Ubuntu 20.04, you need to use the pip command instead of pip2.

bepitama, 2020-05-14:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git python python-tk
sudo apt-get update
sudo pip install --upgrade bCNC
wget http://ports.ubuntu.com/pool/main/p/pyserial/python-serial_3.4-2_all.deb
sudo chmod +x python-serial_3.4-2_all.deb
sudo dpkg -i python-serial_3.4-2_all.deb

DodoLaSaumure, 2020-07-30:
On a fresh install of ubuntu 20.04
I had to install  the following
install pip3

```
sudo apt update
sudo apt install python3-pip
```
install pyserial
`python3 -m pip install pyserial`
run bCNC
`python3 -m bCNC`