vlachoudis/bCNC #1395: No G03 codes in .SVG import.

paulvdhoeven, 2020-04-27:
When opening wings.svg in bCNC 0.9.14-dev: (With Python 3.6 on Linux) all the arcs in a path are read as G02 codes, while 10 of them should be G03.
The file was exported from FreeCAD 0.18.16146
When the file is opened in Xviewer or Inkscape it looks as expected.
The coordinates are OK, When I manually change the G02 codes to G03 in bCNC the path is good.

[wings.svg.zip](https://github.com/vlachoudis/bCNC/files/4541945/wings.svg.zip)

DodoLaSaumure, 2020-05-04:
@paulvdhoeven please let us know if this solved your problem.
Please do not close this issue until it is merged on the head.
Have fun
Pierre

paulvdhoeven, 2020-05-05:
Short: Yes, it works for me.

Long version:
I have some trouble getting to grips with both Python and git. Even cloned the whole DodoLaSAumure fork, which did not work. (Compared with meld, it does not have the 6-axis stuff, some extra exceptions etc). Only then I saw the patch for svg_elements.py. and I added the "<0" on line 3607 with a text editor, and now the import of the .svg file looks as expected.

I am amazed by the amount of code for generating a "simple" arc and how you guys find the right spot to add such a seemingly simple fix.

Harvie, 2020-05-05:
Hey don't close this. Nothing was fixed yet :-)

Harvie, 2020-05-05:
@DodoLaSaumure can you plese make pullrequest? I am aware of #1333, but we need to split this down to multiple pieces and merge separately... It's too much stuff going on at once... especialy because some of the changes i do not agree with yet.

DodoLaSaumure, 2020-05-05:
Thank you guys for your feedbacks, I appreciate it very much
Have fun with bCNC
Pierre

Harvie, 2020-05-05:
Fixed by #1396 thanks

closed 2020-05-05 by Harvie