vlachoudis/bCNC #1397: Set WPOS fails under Python 3

avarakin, 2020-05-06:
Getting this stack trace on mouse click:

Traceback (most recent call last):
  File "/home/alex/.local/lib/python3.8/site-packages/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/home/alex/.local/lib/python3.8/site-packages/bCNC/CNCCanvas.py", line 564, in click
    self.actionWPOS(event.x,event.y)
  File "/home/alex/.local/lib/python3.8/site-packages/bCNC/CNCCanvas.py", line 479, in actionWPOS
    self.app.mcontrol._wcsSet(u,v,w)
TypeError: _wcsSet() missing 3 required positional arguments: 'a', 'b', and 'c'

DodoLaSaumure, 2020-05-08:
@avarakin 
Cannot reproduce with py2 or py3. I am setting WPOS successfully.
Can you please give precisions on what you do exactly to reproduce this bug ?
Thanks
Pierre

DodoLaSaumure, 2020-05-08:
ok got it, i was not working on the head of bCNC.
I take a look.
PIerre

DodoLaSaumure, 2020-05-08:
@avarakin please try commit e7fc199 and tell us.
Have fun
Pierre

PR #1400 (2020-05-09): Update _GenericController.py
fix #1397
Tested with py2 & py3
commit e7fc1997 (2020-05-09): Update _GenericController.py

fix #1397
Tested with py2 & py3
closed 2020-05-09 by Harvie