vlachoudis/bCNC #1399: Serial connected but not...

Antoine8082, 2020-05-08:
Hi, 
Setup:
Dedicated "old" Core 2 duo Laptop, 2Go, ATI.
Step1 HDD : 5400rpm 160Go
Step2 HDD : SSD 60Go 

ESP32 (connexions/LCD) + STM32 (motors control), 4 axis shield on USB or Wifi (AP mode) connexion.

Step1:
Fresh install of Lubuntu and bCNC (Git version following your steps), all work as it should.

Step2:
I have receive the SDD and switch the slow HDD.
Fresh install of Mint (nicer than Lubuntu) and bCNC.
Serial is connected (ESP32 screen reboot at connexion) but, GRBL connexion stay "not connected/red".
$10 from 1 to 10 have no effect...

Same symptoms on a fresh (same) Lubuntu.

It work as it should if I re-plug the old 5400rpm disk with the first Lubuntu.
It work as it should on my Win 10 Surface tablet.
So, an ESP32 side issue is not.

I'm lost, no idea to dig.
Did you ?
Thanks

Antoine8082, 2020-05-10:
[image]

Antoine8082, 2020-05-10:
After further investigation I realised that the version installed on the old HD was a few generations previous to the one installed on the new SSD hard-drive. 
I installed from the commit 23e2611cff44ee36da999a431a4f1332a33de1e1 and set enabled6axis to false.

My setup is currently working. 

I’m in a situation where I only have an A axis, and do not have a B and C axis. Is this a common setup? 

 (in my case word == 5)  so the code fails at 
float(word[5])

Should the lines 
```if len(word) > 4:``` be changed to allow for having an A axis, but not a B or C axis?

axemag, 2020-05-20:
Hi,

I've spent a few evenings debugging this same (or very similar) issue. I had bCNC working together with my GRBL 0.9j CNC router a while ago, but now with a new computer it just would not fully connect. I got the "Not connected" red button, but clicking it said "Connection is established with Grbl".

I found two issues:
**Issue 1**
After connection bCNC expects a status report looking like:
`<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>`
but my CNC router only sent:
`<Idle,MPos:0.000,0.000,0.000>`

Seeing other forum posts suggesting to change GRBL parameter $10 to 1, did not help as that was default in bCNC. Looking at the GRBL documentation it stated that $10 should be set to 3 for both MPos and WPos in the status report. I tried changing the value in bCNC but that did not change anything. After that I connected to the router using a terminal software and there issued the command `$10=3`, which made the difference. bCNC now got the WPos too, after connecting. It did not solve my problem though.

**Issue 2**
In bCNC there is a regular expression parsing the status report for state, mpos and wpos. In commit eb870a6f this regular expression (`STATUSPAT`) was changed to take 6 axes, but from what I can see, it no longer supports 3 axes. Reverting the changes done to 'STATUSPAT' and 'POSPAT' seems to help, I can now connect and get positions from the router. I have not run any CNC jobs yet though.

Maybe @tsmarks has some input here?

```
diff --git a/bCNC/controllers/_GenericController.py b/bCNC/controllers/_GenericController.py
index f5cf6b0..2b1f49c 100644
--- a/bCNC/controllers/_GenericController.py
+++ b/bCNC/controllers/_GenericController.py
@@ -8,8 +8,8 @@ from CNC import CNC, WCS
 import time
 import re
 
-STATUSPAT = re.compile(r"^<(\w*?),MPos:([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),WPos:([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),?(.*)>$")
-POSPAT   = re.compile(r"^\[(...):([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*):?(\d*)\]$")
+STATUSPAT = re.compile(r"^<(\w*?),MPos:([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),WPos:([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),?(.*)>$")
+POSPAT   = re.compile(r"^\[(...):([+\-]?\d*\.\d*),([+\-]?\d*\.\d*),([+\-]?\d*\.\d*):?(\d*)\]$")
 TLOPAT   = re.compile(r"^\[(...):([+\-]?\d*\.\d*)\]$")
 DOLLARPAT = re.compile(r"^\[G\d* .*\]$")
 SPLITPAT  = re.compile(r"[:,]")
```

tsmarks, 2020-05-24:
That is odd.  I tested that with a 3axis system.  Are we going to need some
logic in here to clear this issue?
No insults implied but, is this problem universal or just one user?
Wondering if it is a firmware version acting different than my test unit.

Tim

axemag, 2020-05-25:
My router came pre-loaded with GRBL 0.9j when I bought it. I have never re-flashed it, so I can't guarantee that it wasn't customized by the factory.

Taking one step back, do you agree that the new regexp:s do not match the output from my router?
```
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
```
If so, can it be some setting that differs in my 3-axis system and yours? As you see in my "issue 1" above, it seems my router had lost at least one setting. How do these status report messages look when they come from your test unit?

If we can conclude that our machines output different report messages and that both are valid, one could add optional groups to the regexp, but for reduced code complexity it may be better to use different regexp:s depending on the 6-axis setting.

(Forgive me if something I say makes no sense, I've only spent a few hours digging through the code, trying to solve my specific problem.)

voxelbec, 2020-05-28:
I'm encountering a similar issue.

Serial connection seems to be successful but bCNC remains 'Not Connected'. Can communicate with GRBL via the bCNC terminal and successfully execute manual gcode.
Upgrading from GRBL 0.9 to 1.1 did not resolve. 

Fresh windows10 install.
GRBL 0.9 then 1.1
Python 2.7
bCNC freshly installed via pip

lbattraw, 2020-05-31:
I'd really like to see a little more flexibility in the regexp patterns and state tracking since I'm seeing some of the same issues others have noted, though in my case it's because I've tried using bCNC with grblHAL (Arduino Due) and Grbl_Esp32 (LOLIN D32 PRO).  The ESP32 port is much more obviously broken in terms of backwards compatibility, but grblHAL is frustratingly close to "stock" Arduino UNO/grbl 1.1h.  I can occasionally get it in idle or jog state, however it always gets stuck in that state even though it seems to understand what's going on when you click on the state button.

Anyway, I know that's not much of a concrete goal so I wanted to see if I could help out to get grblHAL functional as it seems much closer to a working state.  It sure is annoying seeing it say it's disconnected however when you click the status it will correctly say it's idle or jogging/etc, though the X/Y/Z coordinates shown never update.  Is there any way to enable showing the full output from grbl in the serial Terminal window?  Often there's quite a bit of traffic that's just invisible.
Is there anyone who could point be to where bCNC parses out and tracks the current state?  I know Python and regexps, but a little help getting started would be really valuable.