vlachoudis/bCNC #1401: run bCNC on RiPi 7" dcreen

beeniedave, 2020-05-11:
hi,  Hope you can help with a very basic question.  I'm new to raspberry pi's & bCNC.  I want to control my new CNC router using a Pi with the official 7" touch screen to save using the expensive laptop in the dusty workshop.  I have a clean Raspbian (buster) install on a Pi3 B  and running BCNC 0.9.14-dev.  I think python is version 2.7.16.  The install of both went fine, and all is working fine on a 17" monitor.  When using the 7" screen the bottom is cut off the control screen so am unable to jog the machine using this.

I've been googling for a week and can't find a fix;  tried  -f  & -g settings but they all seem to make no difference.  Is there a simple guide available that would help me get working on the 7" screen?

thank you for your patience on this noobs question

Dave

[image]

[rasp-ver.txt](https://github.com/vlachoudis/bCNC/files/4611859/rasp-ver.txt)
[image]

LordPato, 2020-05-13:
Having the same issue here with an 5" screen. 
Even though it's not fully ment for that, for final adjustments it would be good to have it as the fixed screen.

There is no way to scroll the controls to get to the bottom ones.

dantiel, 2020-05-30:
i think it must be at least partly possible by changing fonts configuration in bCNC and also set system fonts and dpi, then hide titlebar and autohide system menu of your raspbian