vlachoudis/bCNC #1402: Change Language

Martin3005, 2020-05-12:
Why I can't change the Language to German??? Thanx for your help

DodoLaSaumure, 2020-05-12:
Halo Martin

In CAM Menu, select your language (upper left) and restart program after you are invited.
Works for me in German with the last release
Best regards
mit freundlichen grüßen
Pierre

[image]

Martin3005, 2020-05-12:
Danke!!!!!!! Die Einladung ist zwar nicht erschienen, aber eine teilweise Übersetzung ist jetzt da, nicht alles, aber die wichtigsten Komponenten!!!

DodoLaSaumure, 2020-05-12:
Thank you for your feedback Martin.
Could you please close this issue since it is resolved.
Thanks forwards.
Pierre

closed 2020-05-13 by Martin3005

Harvie, 2020-12-28:
Reportedly fixed in this PR. Please test and close this issue if everything is OK:
https://github.com/vlachoudis/bCNC/pull/1405#issuecomment-751580531