vlachoudis/bCNC #1403: SPI*NDLE command doesn't work

aleontopoulos, 2020-05-13:
When trying to execute the SPINDLE ON command the application throws the following exception:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/bCNC/Utils.py", line 464, in __call__
    return self.func(*args)
  File "/home/pi/.local/lib/python2.7/site-packages/bCNC/__main__.py", line 1217, in cmdExecute
    self.commandExecute()
  File "/home/pi/.local/lib/python2.7/site-packages/bCNC/__main__.py", line 1244, in commandExecute
    self.execute(line)
  File "/home/pi/.local/lib/python2.7/site-packages/bCNC/__main__.py", line 1608, in execute
    self.spindle.set(True)
AttributeError: Application instance has no attribute 'spindle'

aleontopoulos, 2020-05-13:
After looking into this, it appears that within the __main__ execute method it is referring to self.spindle instead of the Page.frames["State"].spindle boolean this will also be a problem with the spindleSpeed integer variable as well.

DodoLaSaumure, 2020-05-13:
Can you please test @aleontopoulos  it seems ok to me.
Thanks
Pierre

aleontopoulos, 2020-05-13:
@DodoLaSaumure yup that looks like it fixed it. Thanks

PR #1404 (2020-05-13): Update __main__.py
fix #1403 (tested by @aleontopoulos)
commit 24b3806f (2020-05-13): Update __main__.py

fix #1403 ?
closed 2020-05-13 by aleontopoulos