vlachoudis/bCNC #1406: Drill tool make bCNC freeze

mitoufflon, 2020-05-17:
Hello,
I'm runing debian 10.
I'want to make a hole in the center of a circle with the drill tool because I don't want to mill it. Every time I tried to use the drill tool with a simple circle in dxf made from inkscape it end with bCNC in idle/freeze state...
I've made some pcbs, cutting and engraving with the other tools of bCNC witout any problem. I'don't know if I'm using the drill tool in the right way because I can't find any documentation on it...?

So far here the ways I'm using it:
* select a circle
* select the drill tool: tcheck drill center only
* make adjustments for the depth
* clic drill
=> **freeze**

* select circle
* select center tool
* select the drill tool: tcheck or untcheck drill center only
* make adjustments for the depth
* clic drill
=>**freeze**

If somebody have a clue... Thank you!

apshu, 2020-05-29:
Hi mitoufflon,
  I might have submitted a duplicate issue #1411 
  Please check your driller input parameters.

mitoufflon, 2020-06-11:
Hello apshu,

Thank you for your answer it seem's working now! 
So to summerize:
* to drill a hole in the center of a circle we need to : 
  * have the correct thickness configuration  in stock/material
  * tcheck _Drill in center only_
  * no _target depth_.
  * put _peck depth_ with a positive value 
  * choose the number of _Dwell(s)_ according to peck depth

* to drill a non-through hole in the center of a circle we need to :
  * have the correct thickness configuration  in stock/material then 
   * tcheck _Drill in center only_
  * give _target depth_ a negative value within the thickness of the material
  * put _peck depth_ with a positive value 
  * choose the number of _Dwell(s)_ according to peck depth

It could be great to update the wiki with those informations in bCNC/wiki/Tools/Drill

If it's possible I could do a Tutorial for that part.

Thank's again apshu !