vlachoudis/bCNC #1407: "Garbage receive MPos" on Status Request with "?"

Jens101, 2020-05-24:
Hello,
I an pretty new to githup. However I stumbled across a issue, which might be worth reporting.

After an update to version 0.9.14.306 bCNC connected at start-up to the controller without clearing the alarm bar to idle. Also bCNC stopped communicating to the controller. Checking the Terminal window I got the following Message:

_Garbage receive MPos: <Idle|MPos:0.00,0.00,0.00,0.00|Bf:31,128|WCO:-733.00,-884.00,-5.00,0.00>
[G54:-733.00,-884.00,-5.00,0.00]
[G55:-890.00,-745.00,-5.00,0.00]
[G56:-636.00,-499.00,-88.00,0.00]
[G57:0.00,0.00,0.00,0.00]
[G58:0.00,0.00,0.00,0.00]
[G59:-561.65,-740.92,-91.75,0.00]
[G28:0.00,0.00,0.00,0.00]
[G30:0.00,0.00,0.00,0.00]
[G92:0.00,0.00,0.00,0.00]_

It seems that the [GRBL1.py](https://github.com/vlachoudis/bCNC/blob/master/bCNC/controllers/GRBL1.py) goes not cope with 4 axis any more.

A downgrade to bCNC 0.9.14.304 solved the issue for the moment for me.

Harvie, 2020-05-31:
Reported to the author of 6 axis support... Hopefully they will figure where the problem lays...
https://github.com/vlachoudis/bCNC/pull/1384

tsmarks, 2020-05-31:
I am looking into it.  What OS are your using? My test laptop running Ubuntu does not have this problem.  And what grbl settings ($$) are set on your machine?  What grbl install are you using?

It appears the controller is reporting a 4th axis position seen here: MPos:0.00,0.00,0.00,0.00
My grbl only reports 3.  If this is a use case, some changes around line 119 to allow the 4th axis with out 5 and 6 might work.  Without being able to reproduce the error, I don't want to toss test code out.  Would you be willing to try experimental code?

Jens101, 2020-06-01:
Thanks a lot. I was not sure if this issue was important enough.
My  bCNC runs on:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
Linux cnc1 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The grbl responds to $I with:
[VER:1.1f5.02/24/2018:]
[OPT:VMH,AVR_644,SPI_SR,SPI_DISP,PANEL,C_AXIS]

To $$ it responds with:
$0=10
$1=25
$2=0
$3=0
$4=1
$5=0
$6=0
$10=3
$11=0.010
$12=0.050
$13=0
$20=1
$21=0
$22=1
$23=3
$24=100.000
$25=2000.000
$26=25
$27=5.000
$30=1000
$31=0
$32=0
$100=79.839
$101=79.974
$102=320.000
$103=10.000
$110=7500.000
$111=7500.000
$112=1000.000
$113=10.000
$120=250.000
$121=250.000
$122=50.000
$123=10.000
$130=750.000
$131=900.000
$132=130.000
$133=10.000

I have no problem to test experimental code. I just would set up a separate SD Card in order to avoid any harm to my existing installation. Maybe I could even learn a bit.

tsmarks, 2020-06-02:
OK.  I think I see the issue.  Your grbl is reporting a "C Axis".  If you change line 119 in GRBL1.py from "if len(word) > 4:" to "if len(word) > 5:" it SHOULD ignore the C Axis report and treat your system like a 3 axis.  IF you installed using pip the file could be in /usr/local/lib/python2.7/dist-packages/bCNC/controllers/. IF you downloaded the zip or "git cloned" it will be in controllers where ever you put the bCNC. 

That really is only a temporary fix. I was unaware of a case where grbl reported only 4 axis.  All I saw was 3 or 6. We really should fix this to allow bCNC to acknowledge the 4 axis report if there is a firmware that reports it... but one issue at a time.  Let's get this working before digging a bigger hole.    

Does anyone else have a 4 axis firmware?  Do we want to work this special case into bCNC or just let it treat it as a 3 axis?

pmchealey, 2020-06-07:
Exactly the same for me.
I'm using python 3.8.3 or 2.7.16 (same problem)
When connected, I receive continuously:
//
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000,0.000>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
Garbage receive MPos: <Alarm|MPos:0.000,0.000,0.000,0.000,0.000|FS:0,0>
//
When I unlock, no more communications...
I try the modification on GRBL.py : no change
I return to the .304 version and all is good.

Thank you for your help

pmchealey, 2020-06-07:
... One more information There is this message in the terminal when I unlock : 
//
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/bCNC/Sender.py", line 777, in serialIO
    elif self.mcontrol.parseLine(line, cline, sline):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/bCNC/controllers/_GenericController.py", line 201, in parseLine
    self.parseBracketSquare(line)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/bCNC/controllers/GRBL1.py", line 218, in parseBracketSquare
    CNC.vars["G92C"] = float(word[6])
IndexError: list index out of range
//

regards,

Jens101, 2020-06-07:
Sorry but I've also bad news. The modification did not show any effect. In GRBL1.py I also changed line 128 to "_CNC.vars["state"] = "Garbage 5 receive %s: %s"%(word[0],line)_" (adding the "5"). However the message stay the same with out the additional "5".
Maybe the the change didn't show any effect, because bCNC loads GRBL1.pyc instead of GRBL1.py. I don't know how to check that.

tsmarks, 2020-06-11:
OK. I see two other lines with the same issue:

lines 169 and 215

if len(word) > 4: 
can be 
if len(word) > 5:

If that doesn't work....
could you add 
print(word)
after 109, so i can see what your grbl is sending.

@pmchealey I noticed your error is slightly different than @Jens101 See jens is reporting four positions "<Idle|MPos:0.00,0.00,0.00,0.00|..." While yours is reporting five.  If you run a $$ do you have a 114, 124, and 134?  It would appear you have a 5axis config?  If my guess correct, you may need to use "len(word) > 6" to get this to work.  

After we get this working, I have a good idea of what the final code should be to fix this the right way.  We should be able to test and commit.  Thank you for working through this.

pmchealey, 2020-06-12:
Thank you Tsmarks everything is working!
I change all for "len(word) > 6" in grbl1.py.
There is still a little problem : no display for A in DRO (stay at 0) when I jog

Jens101, 2020-06-15:
For me it does the trick as well. Thanks. To be perfectly honest, I did not get why. Anyway.
Just for giving a complete feedback here is the console output, resulting from the 'print' inclusion at line 110:

`new-config Utils <ConfigParser.ConfigParser instance at 0x75b9e4e0>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 20971610
Not Loading 6 axis displays
Not Loading 6 axis displays
Controller state changed to: Alarm (Running: False)
['MPos', '0.00', '0.00', '0.00', '0.00']
['Bf', '31', '128']
['WCO', '-718.16', '-742.17', '-128.21', '0.00']
['MPos', '0.00', '0.00', '0.00', '0.00']
['Bf', '31', '128']
['FS', '0', '0']
['Ov', '100', '100', '100']
['MPos', '0.00', '0.00', '0.00', '0.00']
['Bf', '31', '128']
['MPos', '0.00', '0.00', '0.00', '0.00']
['Bf', '31', '128']
['MPos', '0.00', '0.00', '0.00', '0.00']
['Bf', '31', '128']
['MPos', '0.00', '0.00', '0.00', '0.00']
......`

tmbdev, 2020-08-17:
I'm having the same problem. There seem to be a lot of places where the six axis support assumes that anything with more than 3 axes is 6 axes.

I have a 4-axis controller (an ESP32-GRBL board). The same controller comes in 2, 3, 4, and 6 axes, so I think you can't make assumptions like that.

In the short term, these problems can be fixed with code like this (modified from GRBL1.py):

```Python
		if word[0] == "G92":
			CNC.vars["G92X"] = float(word[1])
			CNC.vars["G92Y"] = float(word[2])
			CNC.vars["G92Z"] = float(word[3])
			#if Utils.config.get("bCNC","enable6axis") == "true":
			if len(word) > 4:
				CNC.vars["G92A"] = float(word[4])
			if len(word) > 5:
				CNC.vars["G92B"] = float(word[5])
			if len(word) > 6:
				CNC.vars["G92C"] = float(word[6])
			CNC.vars[word[0]] = word[1:]
			self.master._gUpdate = True
```

I still haven't tracked down all of these problems, but if I can get something that works on my 4-axis controller, I'll send a PR.

An better solution would be something like this:

```Python
axis_names = "X Y Z A B C D E F G".split()
...
for name, value in zip(axis_names, word[1:]):
    CNC.vars["G92"+name] = value
```

Long term, it might be worth thinking about introducing a class to hold a collection of axis values and associated axis names, and methods like "axisobj.stateToVars(CNC.vars, prefix="G92").

(Also, to avoid breaking existing systems with new features like this, it might be worth having some automated test cases that run bCNC against a simulator.)

tmbdev, 2020-08-18:
NB: small PR was incorporated with the above conditionals yesterday. That fixes the immediate problems of not being able to run at all on 4-axis machines.