vlachoudis/bCNC #1410: bCNC stuck into IDLE mode

dantiel, 2020-05-28:
Hi folks,
this is great software but theres this weird bug ocurring to me 90% of time when click the RUN button and the program goes into RUN mode. I mentioned my problem already in https://github.com/vlachoudis/bCNC/issues/1347
however it did not solve updgrading to newer version, or the python drivers i tried. 

Theres three ways how clicks on the RUN button result to me in which the program shows a weird behavoir

* doesnt go correctly into RUN mode, but whole GCODE is run correctly in the background (console attached)
* doesnt go correctly into RUN mode, and only some arbitrary GCODE is run
* goes into RUN mode as normal but gets stuck into IDLe mode after finishing about ~3 blocks.

theres screenshots of the former cases 
[image]
[image]

MARIOBASZ, 2020-05-28:
@dantiel, discard that it is electrical noise: in my case, that was the problem, it almost always came from the spindle, arduino did not respond except disconnect and reconnect

dantiel, 2020-05-28:
> Antworten: vlachoudis/bCNC <reply@reply.github.com>

that can indeed be discarded, i don’t have the issue of needing to reconnect the device, but the process of running gcode just halts. also i connected the spindle to electricity coming from another room, no cables crossing, just to assure

Drachenkaetzchen, 2020-05-28:
>, it almost always came from the spindle, arduino did not respond except disconnect and reconnect 

I've had the same issue, and indeed it was EMI. The "problem" is that for the user, it looks as if bCNC is stalling. Only the serial log revealed that bCNC was indeed sending "?" and never receives a response. This is not reflected in the "Terminal" tab, so the only way to know is to enable serial logging.

The reason why a reconnect works is because connecting the serial on Arduino resets the MCU.

I think the GUI should warn the user if multiple "?" without a response from grbl.

dantiel, 2020-05-28:
So i managed to Log the serial Using `socat`, and indeed it is only outputting `?` continously after the start. But while is `?` is being put out, gcode is still executed fine as normal. Stuck to implement idle state Happen after few time as usual. It Makes No difference however If a spindle is running or disconnected from current. It is Not an EMI issue. Isnt there a way to prevent bcnc from going into idle? Or Waking it Up?

dantiel, 2020-05-29:
if it was possible to omit the response of the grbl, then it could run uninterrupted i think... isnt there a simple way to achieve this? i was wondering why this happens, because earlier it wasnt happen, but i also updated the OS... i tried on both a mac and pi3B, is it possible that its an issue of the control board of my cnc?

MARIOBASZ, 2020-05-29:
did you try another guide instead of bcnc? (ugs for example). Very long usb cable? Try another cable?

dantiel, 2020-05-30:
cool, ugs looks promising. good idea then i can know its not due to python driver issues.. i already replaced the arduino nano on my cnc controller board with newest gbrl version, without effect. 

may there be a another part of control board that can lead to faulty "?" responses?
is there a line i could alter in the source code or configuration and recompile so bCNC keeps on sending the gcode rather than remaining idle?

dantiel, 2020-05-30:
OK, the issues seems solved. it was a USAGE issue, i did  set a feed lower than 1 (and/or too low rate), resulting in G code `f0` in a move cmd, rather than `f0.000001`, i found this thanks to ugs which alerted me that a feed was undefined.

MARIOBASZ, 2020-05-30:
Grbl / bcnc must be reporting error 22

dantiel, 2020-05-30:
ah i see, it was not obvious to me what the errors mean, there were errors also before when it was running fine. i am just beginner of CNCing. thank you for help

MARIOBASZ, 2020-05-30:
spin slow risk of breaking bit, spin fast burns.