vlachoudis/bCNC #1417: Error installing

chepo92, 2020-06-07:
Using pip to install in a clean install of Raspbian

pip2 install --upgrade bCNC

```
Collecting bCNC
Collecting pyserial<=3.0.1 (from bCNC)
  Using cached https://files.pythonhosted.org/packages/42/68/41bab8a5e3e50705f25358f043181242fa197f313aa0476908d9852c8dc4/pyserial-3.0.1.tar.gz
Collecting numpy>=1.12 (from bCNC)
Collecting opencv-python>=2.4 (from bCNC)
  Could not find a version that satisfies the requirement opencv-python>=2.4 (from bCNC) (from versions: )
No matching distribution found for opencv-python>=2.4 (from bCNC)
```

TGit-Tech, 2020-10-29:
I too got same error - 
Collecting bCNC
  Using cached bCNC-0.9.14.308.tar.gz (1.3 MB)
Collecting numpy>=1.12
  Using cached numpy-1.19.3-cp39-cp39-win_amd64.whl (13.3 MB)
Collecting Pillow>=4.0
  Using cached Pillow-8.0.1-cp39-cp39-win_amd64.whl (2.1 MB)
pip : ERROR: Could not find a version that satisfies the requirement opencv-python==4.2.0.32 (from bCNC) (from versions: 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 4.3.0.38, 
4.4.0.40, 4.4.0.42, 4.4.0.44)
At line:1 char:1
+ pip install bCNC
+ ~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (ERROR: Could no...0.42, 4.4.0.44):String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
ERROR: No matching distribution found for opencv-python==4.2.0.32 (from bCNC)

bosd, 2020-11-08:
Same goes for the Python 3 version.
Seems like the referenced opencv version is too old.
Propably caused by #1468 

Workaround
pip3 install opencv
pip3 install bcnc==bcnc==0.9.14.307

Harvie, 2020-11-08:
So what do you suggest?

bosd, 2020-11-08:
running command: pip install opencv-python==
outputs:

''
ERROR: Could not find a version that satisfies the requirement opencv-python==4.2.0.32 (from bCNC==0.9.14.308) (from versions: 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 3.4.11.45, 4.3.0.38, 4.4.0.40, 4.4.0.42, 4.4.0.44, 4.4.0.46)''

So, It seems the problem is with the specified version of Opencv.
4.2.0.32 this one is probably chosen because it is the last version which supports python2.7 according to
https://pypi.org/project/opencv-python/4.2.0.32/

Yet, it seems that version can no longer be installed (as the case Python 2 and V3)
I've browsed trough some releases of OpenCV, could not find one which supports both Python2 and 3.

I know, it has been referenced a couple of times. 
I agree to the folks who want to drop support for python 2.
(My personal experience python3 version of bCNC is that it's good enough for a daily driver)

Python2, might be installed on many systems and may be the default / preffered.
Following the installation instructions, may lead for (not so tech savvy) new users that bCNC is installed using python2.
Leading to installation issues like in #1468 .
"The too difficult too install" may lead users to look for alternatives.

**My suggestion** is to drop python2 support and improve installation experience.
(or archive a python2 version for someone who needs that for any specific reason)

Change setup.py to read
opencv-python>=4.3 

update install instructions as in #1480

Harvie, 2020-11-09:
> (My personal experience python3 version of bCNC is that it's good enough for a daily driver)

I am happy to hear that! I was not really sure, because there seemed to be some issues. i think i saw some bug reports... So lets focus on hunting these bugs... I think we can perhaps drop the support by the new year.

> opencv-python>=4.3

It seems to me that we don't need to drop python2 support just to have latest opencv version. Opencv related code in bCNC was not improved for several years, so it was actualy developed on quite old opencv version and therefore it should work without any issues even with old releases...

> Python2, might be installed on many systems and may be the default / preffered.

I am afraid that users with older OS or without python3 will not be able to install bCNC if we drop the support. And i am even more afraid that this will lead to not-tech-savvy users using old/broken version of bCNC because python2 will not be able to install the latest one and therefere they will be stucked... (some might be stuck even without realizing it!)

bosd, 2020-11-09:
One happy bCNC user here.
Previously used UGS Platform, but encountered loss of serial connection problems. Which cause my lasercutter to burn many sheets.
Now running bCNC on python 3 on large jobs. >1,5MB and >6hour jobs without any problems.
(Running an old version of Deepin OS because on a laptop I had lying around, in the future may use another laptop/distro or raspberry)

Did, some further investigation. 
**On a fresh Rasbian desktop (2020-02-12 Buster)**
``pip install bCNC``
uses python 2 as default.
Seems to install opencv-4.2.0.32 fine.
throws another installation error. Sadly I cannot paste the output here from my virtual machine.

``pip3 install bCNC``
Succesfully installs OpenCV 
``Successfully installed bCNC-0.9.14.308 opencv-python-4.2.0.32

**On Fedora 33: x86_64**
``pip install bCNC``

Install fails. tkinter is not in the standard distribution.
fixed with
``sudo dnf install python3-tkinter``
Could not find a version which satisfies the requirement opencv-python==4.2.0.32

**Possible fix**
Keep the support for now Python 2.7 by hard pointing to opencv-python==4.2.0.32
But point to a newer opencv version for which supports python3.

Local test is working with following code changes in setup.py.
right after the print("Running bCNC setup...") add
```
if sys.version_info[0] >= 3:
	OPENCV = 'opencv-python==4.4.0.46', #use recent opencv for python3
else:
	OPENCV = '4.2.0.32', #defaults to the last python2 supported version of OpenCV
```

and add
``OPENCV,``
in the requirements section

**Possible enhancements**
Use OpenCV not as a hard requirement but as an extra. This it will not be installed on ARM and Arch devices. Or when the user specificly choose to install it as an addon.

bosd, 2020-11-11:
This is also discussed in #1454 
I've coded the conditional formatting based on python version.
Tested and working on Fedora and raspbian.
However I am not so proficient with github yet. So I was unable to commit my changes to that pull request.

Got an error 403. while trying to push my changes to pull request #1454 
I've asked @sttts for write access, or will try to create a new pull request.

Harvie, 2020-11-11:
>  I've asked @sttts for write access, or will try to create a new pull request.

You are supposed to make your fork of repository vlachoudis/bCNC, commit your changes to your fork and then open PR yourself to get your fork merged to main vlachoudis repo

bosd, 2020-11-11:
fixed in commit https://github.com/vlachoudis/bCNC/commit/e939bd108e70fb8456ba544a231e7cd992cd8f84 this issue can be closed

closed 2020-11-11 by Harvie