vlachoudis/bCNC #1418: Add example Gcodes for PCB milling

chepo92, 2020-06-07:
What about adding some example gcodes for testing and tutorials

sle118, 2020-08-10:
Just my two cents here, and for the reference.

There are plenty of how to's all over the Internet, and even if you start with one of the proposed methods, you will always end up developing your own method; you will adapt a workflow that is suitable to your CNC and skills.

Here is a fairy comprehensive write up which I found to be very useful. It also uses a number of tools which are familiar to me (bCNC, FlatCAM, etc).

http://debreuil.com/blog/pcbs/scratch-milling-pcbs-at-home/

chepo92, 2020-08-10:
I know there are plenty of tutorials, I meant an official/original tutorial so this repo is self contained and maintained, third parties tutorials get obsolete, are put down the website, not always complete, etc. 

Probably I would take your two cents and update the wiki :D, thanks

closed 2020-12-10 by chepo92