vlachoudis/bCNC #1419: Feature Request: Add a "probe here"/"set current pos as probe position" button

chepo92, 2020-06-08:
I usually have to manually enter the current position to probe at current location, it would be a nice addition to have a button that sets the current XY position to the probing menu (In the case I know the workpiece is level/ don't want to do autolevel)

MARIOBASZ, 2020-06-08:
No need: delete X and Y fields, only target value in Z

chepo92, 2020-06-08:
> 
> 
> No need: delete X and Y fields, only target value in Z

Nice!, I didn't knew that behaviour, I'll test it, anyway that is not very touch screen and no-keyboard friendly (that's why I asked the button), because I had to enter the XY position every time I wanted to probe, maybe leaving it blank the first time will do the trick, but a custom/hard coded default in blank may work

Any method/config advice to make the fields default to (blank, blank, some_Z_value) ?

MARIOBASZ, 2020-06-08:
I usually have the X and Y fields blank and Z with an objective value, so if you move the tool to a new position, perform a probe and it is done where you were looking for according to your concern

closed 2020-12-10 by chepo92