vlachoudis/bCNC #1421: RPI4: GPIO plugin?

eglische, 2020-06-08:
I'm sorry if this is a dump question, but i am not finding anything by searching, and after giving up to patch RPI to RT for linuxcnc, i am about to go nuts.

Does a mod or plugin exists, to send the step and dir directly to gpio, and also to read some basic inputs like a e-stop?

And if not, why does it not exist? it should be pretty simple to do, and it would help big time with applications like plasma, where a usb connection is iffy... Such thing should be all the rage for DIY projecs like MPCNC.

Cheers
Roger

Harvie, 2020-06-08:
We don't do this, you need extra microcontroller board. Cheapest are like $2 and there are lots of benefits in using it. For me there's no point to use raspi GPIO. I am not saying it is not possible, but it is not supported by bCNC and i don't really like the idea in general.

bCNC is not motion controller. It only sends g-code to motion controller (like GRBL)

If somebody manages to write nice bCNC plugin to stream g-code to LinuxCNC, i think we can accept this plugin to bCNC.

eglische, 2020-06-08:
extra controller board means an additional interface, usually over ethernet or usb.
if you want bcnc to power a plasmacutter or a setup with a frequency drive, it could be easier for the user to implement, as the gpio helps cut out the noise/emi problems that can mess up serial interfaces. both ethernet and usb has more sensitive electronics involved. so if a direct aproach is possible, the user only needs to worry about emi from the powersource, and that the dir/step cables got ferrites on it. with usb in particular more is needed i find.

just as a input.... i have the problem right now, and i am going now for a old pc with parallel port, because it's what works with most setups and it's less problematic.

eglische, 2020-06-08:
but thank you for the answer :-)

closed 2020-06-08 by eglische

Harvie, 2020-06-09:
I think you will run into EMI problems when using raspi GPIO directly as well... and raspi is probably even more sensitive. If you connect arduino using serial line from raspi instead of using usb converter, it will be probably bit more stable. You can even use optocouplers to isolate the serial interface.

If you still want to use GPIO drivers to directly drive the steppers, it should be possible. LinuxCNC has some control commands like `linuxcncrsh` or something (i am not linuxcnc user, so i am not sure which is suitable the most). Maybe it even has some network interface. You can then add new controller plugin to bCNC in this directory, which will connect to it:

https://github.com/vlachoudis/bCNC/tree/master/bCNC/controllers